Ultimaker/Cura #11729: G29 UBL causing heaters to turn off

Sonicboom247, 2022-03-29:
### Application Version

4.13

### Platform

Windows 10 20H2

### Printer

Artillery Sidewinder X1

### Reproduction steps

adding the below to the start gcode is causing both the hotend and bed to reset to 0 at start of print
M75
M117 Homing and Leveling
G29 A ;Activate UBL; 
G29 L0 ;Load Mesh slot 0
;Wait for final heating
G29 J3 ;Probe Bed for mesh tilting

### Actual results

Hotend and Bed temps are set to 0

### Expected results

Hotend and bed should remain set to slicer temps

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[PETG Flow 74calibration cube.zip](https://github.com/Ultimaker/Cura/files/8374251/PETG.Flow.74calibration.cube.zip)

GregValiant, 2022-03-29:
I think something is going on in the firmware.
M140, M190 (bed), M104 M109 (hot end) are the temperature commands.  When the printer sees them, it goes to the indicated temperature.  It will stay there forever until you either unplug the printer, or it sees another temperature command.  That's why M104 S0 and M140 S0 are included at the end of a gcode file.
Within your gcode file - between the temperatures being set, and the actual start of Layer:0 there are no temperature commands.  The print bed should be at 65° and the hot end at 235° until line 814 where they shut off.

Sonicboom247, 2022-03-29:
I have troubleshot this for the last few weeks.  without any G29 commands, the temp is set to 65 and 235 at start, and although no mesh probing is done, the temp is left the same and the print starts just fine.  Also tested in Prusa with the same start gcode including G29 and it works fine.. currently downloading and installing 4.13.1 to see if it changes anything.

GregValiant, 2022-03-29:
But there is nothing in the gcode telling the printer to turn off the hot end and bed and so it must be something in the printer that is causing it.
You can change your StartUp gcode to catch the problem.  (BTW Do you really want the Z moving before you auto-home?)

; PrusaSlicer start gcode for Artillery Sidewinder x1
; Last updated 3112022
M140 S{material_bed_temperature_layer_0} ;Start to heat the bed
M109 S165 ;heat the hot end to 165 and wait for it
M190 S{material_bed_temperature_layer_0} ;wait for the bed to finish heating
G1 Z3 F3000 ; move z up little to prevent scratching of surface
G28 ; home all axes
M75 ;start the print timer
M117 Homing and Leveling ;send a message to the LCD
G29 A ;Activate UBL; 
M420 L0 ;Load Mesh slot 0
;Wait for final heating
G29 J3 ;Probe Bed for mesh tilting
M109 S{material_print_temperature_layer_0} ;Heat the hot end up to print temperature
M190 S{material_bed_temperature_layer_0} ;make sure the bed is at it's set temperature

So the bed and hot end get hot, but the hot end only goes to 165 so it doesn't ooze during leveling.  The printer does whatever it does with the UBL commands and then the hot end goes up to the Initial Layer Print Temperature and the bed is checked to make sure it is up to temp.

Sonicboom247, 2022-03-29:
> But there is nothing in the gcode telling the printer to turn off the hot end and bed 
correct, and if i remove all G29 references, then after the initial warm up, the print starts normally, If i leave them the print still starts normally but temps reset to 0.  funny thing is i can watch it and increase the temp manually either by Octo or the touch screen and it will continue to print just fine....but i see what you are saying, i also removed all M190 and M109 references as well thinking those where the issues but it is still setting them to 0 at start of print.. scratching my head... going to go play some snowrunner to get my head clear..

closed 2022-03-30 by Sonicboom247

Sonicboom247, 2022-03-30:
@GregValiant just want to say thank you, it looks like the Zprobe end script was causing the issue.. Z_PROBE_END_SCRIPT "G1 Z10 F4000\nG1 X0 Y0\nG1 Z10\nM140 S0\nM104 S0

Sonicboom247, 2022-03-31:
Exactly, that's what I get for messing with firmware when i should be sleeping..oh it's all cleaned up now, like you said just a Z move and XY, rest is history