Ultimaker/Cura #11730: The ";LAYER_COUNT:" in a gcode file is off by 19

GregValiant, 2022-03-29:
### Application Version

4.13.1

### Platform

Windows 10 Home (onboard video OpenGL 4.0 in compatibility mode)

### Printer

Creality E3Pro

### Reproduction steps

Sliced a file and checked the Cura Preview - 392 layers.  I used a skirt.
Saved the gcode file.

### Actual results

Open the file in NotePad and see
;LAYER_COUNT:411
;LAYER:0
I was going to drop the temperature at layer 400, but there wasn't a layer 400.
The last layer in the file is LAYER:391 as the preview indicated it would be.

### Expected results

The layer count should be 392.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I have noticed this before.  I am not sure it happens all the time.
The gcode, project and log are included in the zip.
[LayerNumberIssue.zip](https://github.com/Ultimaker/Cura/files/8374512/LayerNumberIssue.zip)

smartavionics, 2022-03-29:
It's including the top support-blocker in the layer count, when I moved that below the top of the model, the LAYER_COUNT value became 392.

fvrmr, 2022-03-31:
Hi @GregValiant thank you for your report.
I will bring it up with the team. Keep you posted.

GregValiant, 2024-11-14:
I checked and 5.8.0 does not have the problem.  I'll go ahead and close this.

closed 2024-11-14 by GregValiant