Ultimaker/Cura #11733: Slicing Incorrectly

Canon182, 2022-03-29:
### Application Version

4.13

### Platform

macOS Monterey

### Printer

Ender 3

### Reproduction steps

Put my part in Cura
Sliced them
Pressed "Preview"

### Actual results

There was a hole in my part after pressing "Preview

[image]
"

### Expected results

The part to be whole with no missing parts and ready for printing

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

In the picture the red and yellow are suppose to be connected
*Its part of a helmet*

[image]

Here is the project

https://www.thingiverse.com/thing:4757926/files

GregValiant, 2022-03-30:
Load the model in Cura.  Select your settings.  In the "File" menu select "Save Project".  Zip the file and post it here.

Canon182, 2022-03-31:
[CE3_Main.3mf.zip](https://github.com/Ultimaker/Cura/files/8393462/CE3_Main.3mf.zip)

GregValiant, 2022-04-01:
When the warning comes up "The model is not watertight and may not print properly" then there are problems with the model that will cause it to not slice properly either.  If you turn on "x-ray view" you will see that one entire surface is red indicating open seams.  The model needs repair.
[image]

Canon182, 2022-04-01:
I cut the model in half so it could fit on my printer. I don't know how to fix the red, so if you could please tell me how that would be great. Thnx a lot for your help

GregValiant, 2022-04-01:
You could try the software you sectioned it with, or MS 3D Builder, or Blender, maybe the Mesh Tools plugin in Cura could do it.  That entire red area is open and so Cura can't tell what is the inside and what is the outside.

Canon182, 2022-04-06:
I downloaded the mesh tool and tried using it to fix the part. Ive never used it before so I don't know if I didn't use it correctly or not.

fieldOfView, 2022-04-06:
The Mesh Tools plugin is limited in what it can do; it doesn't do magic. This might just be a case where "Fix simple holes" fixes it. If it doesn't, you'll need a different software to repair the model, like Microsoft 3d Builder, which comes with Windows 10, or Meshmixer (which is free). Blender is also free, but not an easy tool to get to know.

Canon182, 2022-04-06:
Ok cool. I also have access to Solidworks, however its not mine and I cant use it until Monday. I think it might work, but I have to wait and I would prefer to get the print started sooner. (Not like Im on a deadline or anything...It's just a long print). I also have blender but have no idea how to use it... but my sister might so I could ask her. Also Im using a mac so I might be limited to what software I can use

GregValiant, 2024-11-15:
I'll go ahead and close this as it was no a Cura bug.

closed 2024-11-15 by GregValiant