Ultimaker/Cura #11742: Time Estimate for Each Layer

rachael7, 2022-03-30:
### Is your feature request related to a problem?

I recently had a print fail due to what appeared to be too short layer times causing heat buildup and slumping (part was in ABS with almost no fan). Diagnosing the print failure led me to want to check the time for each layer, but that information was not available in Cura. On this particular part, the cross-section was relatively constant, so I could get a good estimate by simply dividing total time by the number of layers; but on parts with large variations in cross-section, that technique would not work.

### Describe the solution you'd like

I would like the preview window to show the estimated time for the layer that the slider is on (the currently displayed layer). A logical location would be near the scrubbing bar at the bottom of the screen (where the play/pause button is). The bar could be labeled 0s at the beginning of the bar and have the total time for the layer on the right side of the bar. An idea implementation would show the time to the position of the slider as well, much the way the scrubbing controls work on a video player or editor.

### Describe alternatives you've considered

On the particular part that raised this concern for me, the cross-section was relatively constant, so I could get a good estimate by simply dividing total time by the number of layers; but on parts with large variations in cross-section, that technique would not work.

### Affected users and/or printers

This will provide valuable information for anyone printing small parts who is concerned with the minimum layer time and related functions. It would allow tuning of print speed, fan speed, and other parameters based on the time for each layer.

Flying-Roger, 2022-12-25:
Good idea !
Monitoring the progress of the printing on a webcam, I can easily match a layer position in the Cura preview. Having a time associated with the layer would allow me to get an estimate of the remaining time. Of course, this will be approximate, but as the Cura total time estimates are pretty good since I tuned the accelerations, this would be good enough. Ideally, this should be integrated in the Cura preview. 
Otherwise, maybe a pluging would do it.

BarryTice, 2023-06-09:
I came here to make this same request. I'm experimenting with hot-swapping filament colors mid-model, pushing a new color down my Bowden tube against the end of the previous color, and will need to be at the printer ready to make the change right on schedule — if only I knew what the schedule was. Being able to see that layers 1 through 8 will take 26 minutes each means I know I should show up by 50 minutes in if I want a change at the end of layer 2.

FlashMoonUK, 2024-02-14:
Has this ever been added? if so I can't find it, Its a very useful feature in other slicers.

Rolleroo, 2024-06-02:
If someone knows of a way to estimate layer height time, even roughly, that would be great. My Cura estimates tend to be bang on so I think it would be pretty accurate.

endless-iterations, 2026-09-05:
Here you go everyone

https://github.com/endless-iterations/Layer-Timing-Info-Plugin

PartySausage, 2026-09-05:
Neat 😊

PartySausage, 2026-09-05:
There is a bug in your plugin. If the sliced model on the Buildplate is positioned in the region that will be occupied by the timing info data & play button; When you switch to the Preview, Cura crashes with the following log entries & Crash Report Data

```
2026-09-05 23:24:45,243 - DEBUG - [MainThread] UM.Controller.setActiveStage [222]: Setting active stage to PrepareStage
2026-09-05 23:24:45,274 - DEBUG - [MainThread] SidebarGUIPlugin.SidebarGUIProxy.setActiveView [132]: Setting active view to SolidView
2026-09-05 23:24:45,284 - DEBUG - [MainThread] UM.Controller._updateActiveView [150]: Setting active view to SolidView
2026-09-05 23:24:45,295 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [62]: Loading [C:\Program Files\UltiMaker Cura 5.13.0\share\cura\resources\shaders\xray.shader]...
2026-09-05 23:24:51,902 - DEBUG - [MainThread] UM.Controller.setActiveStage [222]: Setting active stage to PreviewStage
2026-09-05 23:24:51,902 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2026-09-05 23:24:51,902 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: TypeError: unable to convert a Python 'QQuickItem' object to a C++ 'QObject*' instance
2026-09-05 23:24:52,027 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2026-09-05 23:24:52,027 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: TypeError: unable to convert a Python 'QQuickItem' object to a C++ 'QObject*' instance
2026-09-05 23:24:52,043 - DEBUG - [MainThread] SidebarGUIPlugin.SidebarGUIProxy.setActiveView [132]: Setting active view to SimulationView
2026-09-05 23:24:52,046 - DEBUG - [MainThread] UM.Controller._updateActiveView [150]: Setting active view to SimulationView
2026-09-05 23:24:52,172 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2026-09-05 23:24:52,173 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: TypeError: unable to convert a Python 'QQuickItem' object to a C++ 'QObject*' instance
2026-09-05 23:24:52,467 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2026-09-05 23:24:52,468 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: TypeError: unable to convert a Python 'QQuickItem' object to a C++ 'QObject*' instance
2026-09-05 23:24:52,659 - WARNING - [MainThread] cura.UI.PrintInformation.setBaseName [424]: Unsupported Mime Type Database file extension Cube
2026-09-05 23:24:52,662 - WARNING - [MainThread] cura.UI.PrintInformation.setBaseName [424]: Unsupported Mime Type Database file extension 
```

```
Thread 0x0000141c (most recent call first):
  File "C:\Program Files\UltiMaker Cura 5.13.0\share\cura\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 1010 in run
  File "threading.py", line 1073 in _bootstrap_inner
  File "threading.py", line 1030 in _bootstrap

Thread 0x00005b04 (most recent call first):
  File "threading.py", line 355 in wait
  File "threading.py", line 507 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1073 in _bootstrap_inner
  File "threading.py", line 1030 in _bootstrap

Thread 0x00000a8c (most recent call first):
  File "threading.py", line 355 in wait
  File "threading.py", line 507 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1073 in _bootstrap_inner
  File "threading.py", line 1030 in _bootstrap

Thread 0x00006880 (most recent call first):
  File "threading.py", line 355 in wait
  File "threading.py", line 507 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1073 in _bootstrap_inner
  File "threading.py", line 1030 in _bootstrap

Thread 0x00004374 (most recent call first):
  File "threading.py", line 355 in wait
  File "threading.py", line 507 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1073 in _bootstrap_inner
  File "threading.py", line 1030 in _bootstrap

Current thread 0x00003ee8 (most recent call first):
  File "cura\CrashHandler.py", line 397 in _logInfoWidget
  File "cura\CrashHandler.py", line 182 in _createDialog
  File "cura\CrashHandler.py", line 86 in __init__
  File "cura_app.py", line 169 in exceptHook
  File "C:\Users\graem\AppData\Roaming\cura\5.13\plugins\LayerTimingPlugin\LayerTimingPlugin\LayerTimingPlugin.py", line 172 in _attachHUDToStage
  File "cura\CrashHandler.py", line 460 in _show
  File "UM\Event.py", line 218 in call
  File "UM\Qt\QtApplication.py", line 510 in event
  File "cura\CuraApplication.py", line 1232 in event
  File "cura\CrashHandler.py", line 460 in _show
  File "UM\Event.py", line 218 in call
  File "UM\Qt\QtApplication.py", line 510 in event
  File "cura\CuraApplication.py", line 1232 in event
  File "cura\CrashHandler.py", line 460 in _show
  File "UM\Event.py", line 218 in call
  File "UM\Qt\QtApplication.py", line 510 in event
  File "cura\CuraApplication.py", line 1232 in event
  File "UM\Qt\QtApplication.py", line 421 in exec
  File "cura\CuraApplication.py", line 961 in run
  File "cura_app.py", line 243 in <module>
```

endless-iterations, 2026-09-05:
Thank you for letting me know AND sending the debug logs, can't reproduce on my send.  I'll get on it.

PartySausage, 2026-09-06:
In the 1.0.1 version the data doesn't display. The release notes suggest that there were some QML file changes but the LayerTimingHUD.qml file in 1.0.1 is the same as 1.0.0

PartySausage, 2026-09-08:
I've created a version of your Plugin, which resolves the crash and can be expanded / collapsed as required.  Rather than anchoring to the Play Button & Layer Slider it's centred on the Horizontal Path Slider and only uses that as a reference point. This was a really good learning exercise so thanks for the innovation. 

The plugin is still fundamentally your design, I wouldn't have considered (or been competent enough) to create it myself and ownership still rests with yourself.  If you would prefer this version removed from here, let me know & I will do so.  

[LayerTimingPlugin_HD.zip](https://github.com/user-attachments/files/31947422/LayerTimingPlugin_HD.zip)

[image]

[image]

karabas2011, 2026-09-09:
> In the 1.0.1 version the data doesn't display. The release notes suggest that there were some QML file changes but the LayerTimingHUD.qml file in 1.0.1 is the same as 1.0.0

The same. I use 1.0.0 now HD version

PartySausage, 2026-09-11:
If you are interested, I've created a new version, which is a bit more efficient & resolves some alignment / layout glitches. It now has two layout options - Normal / Compact

[image]

[image]

[LayerTimingPlugin_HD.zip](https://github.com/user-attachments/files/32131588/LayerTimingPlugin_HD.zip)

endless-iterations, 2026-09-25:
Sorry for the delay, I was off grid working a gig.  You took this from a rough cut to a polished gem!  I'm probably going to shut the releases down and just hyperlink back here because this is the bees knees!

endless-iterations, 2026-09-25:
> > In the 1.0.1 version the data doesn't display. The release notes suggest that there were some QML file changes but the LayerTimingHUD.qml file in 1.0.1 is the same as 1.0.0
> 
> The same. I use 1.0.0 now HD version

Yeah that was definitely my noob error.  When I put it together for myself it was just the single file I dropped in the folder.  Not well versed the ins and outs of packaging but I thought it would be more appealing to the everyday user.  

Was it the plugin placement being linked to the slider when it should have been linked to the slider parent that was causing the crash on your end?

endless-iterations, 2026-09-25:
[image]

PartySausage, 2026-09-25:
> <img alt="Image" width="997" height="365" src="https://private-user-images.githubusercontent.com/141884264/659111848-d7b229d4-4a3f-4a92-92f7-aef31905a3c1.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3OTAzNjQ2MTUsIm5iZiI6MTc5MDM2NDMxNSwicGF0aCI6Ii8xNDE4ODQyNjQvNjU5MTExODQ4LWQ3YjIyOWQ0LTRhM2YtNGE5Mi05MmY3LWFlZjMxOTA1YTNjMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjYwOTI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI2MDkyNVQxOTI1MTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMTI5ODYzZjM2ODQyYzhkYWU1ZWI5NzZlMWVlOGM0OGZjMjczMDM2MWQ5MDEwMzcxNTM0ODU4NGUwZTc4ZWIyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZyZXNwb25zZS1jb250ZW50LXR5cGU9aW1hZ2UlMkZwbmcifQ.ZLXOJyWO3xOA_hTw1cwTobTuJFMZEBK_p446GJO2n8c">

This is my final version that added an animation to the open/closing of the timing panel. Tested with 5.13 & 5.14 Alpha

[LayerTimingPlugin_HD.zip](https://github.com/user-attachments/files/32668091/LayerTimingPlugin_HD.zip)

PartySausage, 2026-10-06:
I thought I was done with ironing out the gremlins in this plugin, but whist I was testing some other plugins I noticed that thie Layer Timing was all over the place when printing in One-at-a-Time mode.
I've made an attempt at fixing it by trying to identify the layers that are attributed to each individual model, which appears to work but needs a wider test.
There are some caveats with both One-at-a-Time & All-at-Once modes in that if you are using any Post-Processing scripts or Plugins that perform layer modifications, then when the file is saved the estimated times in the preview may produce some spuroius values. This is because the Estimated Times in Post-Processed G-Code are still based on the times when the model was sliced and are not updated as a result of any post processing operations.
Anyway have a play & let me know if you stumble on any issues

[LayerTimingPlugin_HD.zip](https://github.com/user-attachments/files/33124724/LayerTimingPlugin_HD.zip)