Ultimaker/Cura #11743: Feature request: Minimum/maximum line width setting

tastyratz, 2022-03-31:
### Is your feature request related to a problem?

The center line in a print is never perfectly sized. Let's use a cylinder as an example. Set your line width too wide and it's skipped Print thin lines "everywhere" and you're going to have a really strange zig zag that is asking for an enormous amount of time, noise, surface finish issues and possible head buildup.

### Describe the solution you'd like

What if you were able to set a minimum and maximum line width and have an adaptive line width setting similar to how the adaptive layer height setting works?

Nozzle pressure changes could be a consideration, but, an internal minimum total length could constrain rapid line width changes.

Let's say we have a configured line width of 0.8 on a 0.6 nozzle and set our max to 0.85 and min to 0.5 (pushing the limits of an e3d nozzle). 
In the example cylinder mentioned above, if there is a 0.71mm center line gap we would normally miss it leaving a hole/double cylinder. With the new settings, cura can print the final line with a line width of 0.71mm making for an exact fit and connectin the solid part.
Consequently, if the center has a 0.05mm gap in it, the center line can scale from 0.8 to 0.85 to fit.
In addition, if a lot of fine details are detected with monotonic order then the inside line could utilize minimum line width to try and better handle the smaller details.
In absense of nearby features or gaps the nominal base line width is selected.

If we want to print wide layers for fast strong prints while having better details, we can.
This could also go a long way towards dimensional accuracy for complimenting horizontal expansion which currently seems to only vary at most a single line width.

### Describe alternatives you've considered

Compromising line width for details and accuracy

### Affected users and/or printers

Everyone could benefit from this.

Asterchades, 2022-03-31:
Good news! This is already facilitated with Arachne which, at least so far as I know the plan to be, should be in Cura 5.

I don't believe you can actually specify a fixed value for the minimum and maximum, but there are settings to choose at what percentage of line width it should choose to use more, thinner lines or less, thicker lines in order to prevent a gap formed where a whole line would otherwise not fit. But the end result should be just about the same in terms of gap fill.

You'll still need to adjust the base line width to print wider lines in general, however.

tastyratz, 2022-05-23:
This can be closed now since Arachne has the functionality! I look forward to testing it out more!

closed 2022-05-23 by tastyratz