Ultimaker/Cura #11751: Retraction speed limited to Initial Layer speed

balloonhotair, 2022-04-01:
### Application Version

4.13.1

### Platform

Windows

### Printer

Qidi

### Reproduction steps

Create gcode for any file

### Actual results

Retraction/detraction speeds are being limited to what the initial layer speed setting is.

### Expected results

Retraction/detraction speeds should be unaffected by the initial layer speed feed rate. Here is a a short piece of gcode to explain. In this example initial layer speed is 20mm/s and  retraction speeds are 40mm/s. The code shows just 20mm/s for retraction and detraction.
G1 F1200 X190.348 Y139.8 E360.41877
G0 F1800 X184.475 Y101.673
G0 X142.996 Y82.059
G0 X104.781 Y107.761
G0 X109.213 Y151.494
G0 X149.28 Y168.45
G0 X179.874 Y137.191
G0 X179.812 Y128.986
G0 X167.742 Y113.452
G0 X152.002 Y96.6
G1 F1200 X149.688 Y96.6 E360.67275

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

See above.

GregValiant, 2022-04-01:
Retractions are always on their own line in the gcode.  There aren't any retractions in your snippet.  The first line is an extrusion at 20mm/sec.  Then there are 9 combing/travel moves at 30mm/sec, and finally another extrusion at 20mm/sec.
A retract and un-retract section would look something like
G1 X37.098 Y35.358 E82.35712  <-- Extrusion
G1 F2100 E77.35712  <-- Retraction
G0 X104.498 Y137.653  <-- Travel
G1 F2100 E82.35712  <-- Un-retract
G1 F4500 X105.134 Y137.184 E82.38569  <--Extrusion

GregValiant, 2024-11-15:
I'll go ahead and close this as this doesn't show a bug.