Ultimaker/Cura #11752: Wall speed does not seem to be used

ysu2, 2022-04-02:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Creality Ender-6

### Reproduction steps

I've noticed that walls seem to print rather slower than what I set.

So I've increased the speed of walls to 120mm/s, but the walls seem to be printed at around 50mm/s.
That is, except for the top & bottom - which is kind of the opposite of what I wanted to achieve.  And I seem to have no control over the wall speed.

Another thing is the infills; they are printed slower than I'd expect, too.  Not sure what controls the speed of this, though.

### Actual results

slow printing of walls & infill

### Expected results

fast printing of walls - as set.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Here's the project:
https://www.dropbox.com/s/8grswunufvdnxtv/TestCube-h2.065.3mf?dl=1

And a video showing the issue:
https://www.dropbox.com/s/xvjdf42z82k2wd2/20220402_110137.mp4?dl=0

The initial layers seem ok, and at around 1:50, after the first 4 layers (the "bottom") it shows a slowdown.

GregValiant, 2022-04-02:
Under "Cooling" you'll find "Minimum Layer Time" and the related "Minimum Speed"..

ysu2, 2022-04-02:
Shoot.
Thanks Greg, and I do apologise for wasting your time - indeed, that fixed my issue.

closed 2022-04-02 by ysu2