Ultimaker/Cura #11758: Object Print Order

Leictreonaic, 2022-04-02:
### Is your feature request related to a problem?

Cannot change print order of imported objects after they've been imported.

### Describe the solution you'd like

Reorder objects in the object list for print order.

### Describe alternatives you've considered

Grouping or importing in the order wanted.

### Affected users and/or printers

Everyone using Cura 4.13.1

fvrmr, 2022-04-04:
Hi @Leictreonaic thank you for your request.
We have a familiar request on our backlog. Which is linked to https://github.com/Ultimaker/Cura/issues/4835 where the request is for printing one at a time. 
I will check with our UX designer if we could add this to the scope of the ticket.

mikebzh44, 2023-02-08:
+1

I was using Simplify3D and it was so easy to change order in the print list !

Why not add some "move up / move down" item on the menu when clicking in the object list :

https://imgur.com/RKRJoDA