Ultimaker/Cura #11759: Drag Select

Leictreonaic, 2022-04-02:
### Is your feature request related to a problem?

Cannot select object by dragging the mouse with left click.

### Describe the solution you'd like

Enable left click drag select to select objects on the build plate.

### Describe alternatives you've considered

Control click one by one.

### Affected users and/or printers

Everyone using Cura 4.13.1

fvrmr, 2022-04-07:
Hi @Leictreonaic thank you for your report.
I will bring it up with the team. Keep you posted!

LukasExemplar, 2024-08-02:
I would like to second the need for this feature, but also contribute my own suggestions:
Have both a box select, circle select (kind of like a brush, scalable) and lasso select option, exactly like Blender does it.

Is it okay for me to post this here? Or do I need to add my own post instead?

CrazedCrusade, 2025-03-17:
Any updates on this? I could really use this feature