Ultimaker/Cura #11760: Concentric bottom has stuttering created in a simple model - 1.0mm nozzle

NilsRo, 2022-04-03:
### Application Version

4.13.0

### Platform

Windows 11

### Printer

Anycubic Mega S

### Reproduction steps

1. slice the project attached

[image]

### Actual results

Stuttering in bottom layer with concentric settings

### Expected results

No stuttering as it matches the size of the nozzle. Perhaps there is a calculation that have some rounding issues?

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[LBoxx Einsätz Sägen.5.zip](https://github.com/Ultimaker/Cura/files/8403973/LBoxx.Einsatz.Sagen.5.zip)

smartavionics, 2022-04-03:
With the current Cura releases to avoid that stuttering, there needs to be an even number of yellow lines. So try reducing the skin line width to 0.8 (or thereabouts). Cura 5 will solve this problem. Hope this helps.

NilsRo, 2022-04-03:
> With the current Cura releases to avoid that stuttering, there needs to be an even number of yellow lines. So try reducing the skin line width to 0.8 (or thereabouts). Cura 5 will solve this problem. Hope this helps.

If Cura 5 heals it it fine. Simply printed lines but looking forward to the fix.

closed 2022-04-21 by NilsRo