Ultimaker/Cura #11761: slicing error

mringersoll, 2022-04-03:
### Application Version

4.13

### Platform

windows 10

### Printer

Ender3

### Reproduction steps

tried to slice a project

### Actual results

unexpected error while slicing and than cura stops

### Expected results

For project to slice does this on all the projects i try

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3_L WIng 2.zip](https://github.com/Ultimaker/Cura/files/8404307/CE3_L.WIng.2.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/8404333/cura.log)

GregValiant, 2022-04-04:
I didn't have any problem with it slicing for my Ender 3 Pro using 4.13.1 on Windows 10 Home.
If you pass along more about your computer setup (motherboard, processor, on a network, multiple monitors, video system) it may help.

mringersoll, 2022-04-04:
Sorry using cura archane beta or however it is spelled it works sometimes and than it don't work and can be slicing the same object MikeSent from my Verizon, Samsung Galaxy smartphone

fvrmr, 2022-04-08:
Hi @mringersoll thank you for your report.
I get indeed a slicing error in Arachne. But slicing it in Master it slices successfully, so it will be fixed in the next release. Since the Arachne Engine will be in next release!
I will close this issue since it will be fixed!

closed 2022-04-08 by fvrmr