Ultimaker/Cura #11766: Cura slicing add strange things

pierreblavy2, 2022-04-04:
### Application Version

4.13.1

### Platform

fedora 35

### Printer

CR10 s pro V2

### Reproduction steps

Open project
Slice
Show preview

See screeshot to see the part that are added during slicing

### Actual results

There is a strange planed, added by the slicing procedure, that's not supposed to be here. This plane is marked WTF on the screenshot

### Expected results

Slicing a tube, should produce a tube

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CCR10SPRO_drawer-tube.3mf.zip](https://github.com/Ultimaker/Cura/files/8410113/CCR10SPRO_drawer-tube.3mf.zip)
[image]

pierreblavy2, 2022-04-04:
**- Exclusive works for me. Thanks :)**
- Inclusive makes things worse

fvrmr, 2022-04-05:
Hi @pierreblavy2 thank you for your report. 
This looks like a duplicate of https://github.com/Ultimaker/Cura/issues/9979
We have a ticket on our backlog for this! I will close this issue, since it is a duplicate.

closed 2022-04-05 by fvrmr