Ultimaker/Cura #11769: Tree support not generating with only touching buildplate

Silithas, 2022-04-04:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Ender 3 pro custom marlin firmware

### Reproduction steps

Reproduction is provided in pictures and files to use the same model and support settings

### Actual results

No tree support appears when even setting branch to 80 degree branching

### Expected results

It generates supports for the chin for a face on a bust and branches down to the bed.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

https://drive.google.com/file/d/1ww-Lji2qsapeoR9Ht3dTCyIrRC_Nhcs0/view?usp=sharing

[image]
[image]

fieldOfView, 2022-04-05:
Have you considered _not_ using the "only touching buildplate" option? That option mostly makes sense for normal supports (as opposed to tree supports). It actually works the same for both types of supports (ie: areas that are not directly above the buildplate, such as the chin, don't get supports).

Asterchades, 2022-04-05:
That actually explains a lot about the (mis)behaviour of tree supports that I've seen in the past. Honestly it also seems counter-intuitive given the nature of tree supports - there's a not-unreasonable expectation that they'd be able to "branch" out over a lower portion of the model to support something, rather than having to potentially mark the existing surface. It gets even more confusing when you consider there are situations where it does reach out that way. If it's not supposed to, by design, that would explain the inconsistency in what gets supported or not.

In either case there's a fault here. Either they're reaching across when they shouldn't or they aren't reaching across when they should. In some models you can see both in effect at the same time.

DerGenaue, 2022-06-06:
You can try this [Reimplemented version of Tree Supports](https://github.com/ThomasRahm/CuraEngine) in the meantime btw