Ultimaker/Cura #11770: sometimes can't slice giving bug error

hiskynet, 2022-04-05:
### Application Version

4.13.1

### Platform

Windows 11

### Printer

Geeetech A10M

### Reproduction steps

give slice command sometimes can't slice and giving bug error
[3dtouch installed]

### Actual results

and if tried to print just ran into regtingles around the plat (Z 500) and extrude even temp is low

### Expected results

1- to print normally 
2- Dual extruder and color mixing plugin not working on Cura

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I'm not pro, but learning

Updated to Marlin 2+ 

tweaked the configuration.h file, auto bed leveling is working, and calibrated and tested on Marlin 1.1.8. but this bug showed, and can't print anymore after updated.

Uncommented

#define USE_XMAX_PLUG //Check
#define USE_YMAX_PLUG

#define ENDSTOP_INTERRUPTS_FEATURE

#define NOZZLE_TO_PROBE_OFFSET { -37, 0, -2.4 }

#define Z_SAFE_HOMING  //3Dtouch

fvrmr, 2022-04-08:
Hi @hiskynet thank you for your report.
Could you share a project file (file - save project) that doesn't slice here in a .zip file?

hiskynet, 2022-04-08:
Thanks a lot for your kind response

I think Cura plugin problem (I guess "universal settings"" plugin)

Best Regards
DrMu

closed 2022-06-15 by nallath