Ultimaker/Cura #11775: [BUG] Slicing supports in diagonal way are too close to object

tispokes, 2022-04-06:
### Application Version

4.13.1

### Platform

W11

### Printer

SWX1

### Reproduction steps

Layer height: 0.12 mm
Support X/Y distance: 0.8 mm
Support Z distance: 0.24 mm
Generate supports in Zig Zag from buildplatform

### Actual results

Supports are too close in diagonal

### Expected results

More distance

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

project zip: https://www.dropbox.com/s/mb8pa8pu2yetvwp/ohmeiEgg.zip?dl=0
[image]
[image]

tispokes, 2022-04-06:
Thanks for the fast reply. I tried X/Y over Z and vis a versa. "Minimum Support X/Y distance" was the last factor I testet before addind the project file. Was 0.4 mm initial.

Chainging "Minimum Support Floor Area" (or all) to 2.0 mm² indeed to something positive, but the diagonal issue still exists. Both Z over X/Y and X/Y over Z tested.
See pictures.
[image]

[image]

GregValiant, 2022-04-06:
I don't get that area when I slice it.  I wonder if it's a video display thing.  (My laptop doesn't support OpenGL 4.1 and so I am in compatible mode).
Have you tried slicing, clearing the build plate, and then opening the Gcode file in Cura?  The gcode reader is separate from the slicer and sometimes things will appear differently.

tispokes, 2022-04-06:
Same on the opened gcode. 
[image]

tispokes, 2022-04-06:
Changed only the initial. Same problem. 
Used now the standard profile "Super Quality" same effect/problem.
Same as Dynamic and Standard.

[image]

GregValiant, 2022-04-06:
Instead of altering the Initial Layer Height I've moved the model down to 1.92 above the build plate and made the other changes I talked about.
[GVohmeiEgg.zip](https://github.com/Ultimaker/Cura/files/8427659/GVohmeiEgg.zip)

tispokes, 2022-04-06:
I did also set the model also to a height which has 0 remainder to the layer height. All my pictures from the second post on are with 0.84 z.
But will try later thanks

tispokes, 2022-04-08:
Opened your 3mf, imported as model with new profile for the settings. remains the same. 
[image]

GregValiant, 2024-11-15:
Is this still an issue with recent versions (5.8.0 and up)?  Can I close this?

closed 2024-12-09 by github-actions[bot]