Ultimaker/Cura #11778: Merge "Settings" and "Preferences" in menu bar

aldencolerain, 2022-04-06:
### Is your feature request related to a problem?

Having two menu items, settings and preferences is confusing.  They have similar meaning and new users frequently click back and forth between the two attempting to find settings/preferences.  When you click on preferences there is only one menu item "Configure Cure" which when opened actually also contains print settings.  When you click settings then "Configure setting visibility" this actually also opens the "preferences" window just in a different tab.  Also worth noting there are multiple ways to change the printer and the filament.

### Describe the solution you'd like

I think these two menus need to be merged and could also be organized a little better.  I think cleaning these menus up would reduce some cognitive load for new users and increase consistency between the menu bar and the Cura preferences and settings window.

### Describe alternatives you've considered

The preference window once open is more intuitive and easy to navigate, perhaps aligning the menu items a little closer to the preferences window tab would help the UI and be more intuitive.

### Affected users and/or printers

Everyone

GregValiant, 2022-04-07:
There is some redundancy in there.  When I first started using the program I was thinking "man, there has to be something different between these".  It turned out that there wasn't.
This is essentially a combine and re-order.  There are only a couple of follow up forms involved.

- Settings and Preferences
    - Active Printer (_MyPrinter.Name_)
        - Machine Settings
        - Extruder Settings
            - Extruder1 (T0)
            - Extruder2 (T1)
            - Etc.
    - All Printers
        - Installed Network Enabled Printers
        - Installed Local Printers
        - Manage Printers
 .............................................
    - Cura Preferences
        - General
        - Settings Visibility
        - Materials
        - Profiles

fieldOfView, 2022-04-07:
In Cura, "Settings" are what you change to affect slicing. The "settings" menu mainly deals with selecting profiles (in the sense that materials and variants are also profiles).

"Preferences" affect how the application behaves, and what configurations are available in your Cura setup.

The problem with just moving that single "Configure Cura" item into the Settings menu is that there will be many support calls from people asking where the Preferences have gone. Also Mac OS is rather stubborn in where it places the preferences menu item (in that it will literally rip out the menu item whereever you place it and put it in a special system-provided menu).

GregValiant, 2022-04-07:
"_Also Mac OS is rather stubborn in where it places the preferences menu item (in that it will literally rip out the menu item wherever you place it and put it in a special system-provided menu)._" - Sometimes I forget (or simply don't understand) the Multiple Operating System aspects.
What it seems is that:
"Preferences - Configure Cura" brings up the "Preferences" dialog with the General tab active.
"Settings - Configure Setting Visibility" brings up the "Preferences" dialog with the Settings tab active.
"Settings - Printer - Manage Printers" brings up the "Preferences" dialog with the Printer tab active.

I guess those are the ones that really confused me.  It just seemed that whichever I selected I always ended up in the same place.

aldencolerain, 2022-04-13:
@fieldOfView I understand the concern about making changes that disorient experienced users and that Mac has special handling for preferences.

If merging the two isn't an option for the reasons you mention, perhaps we can expand the preference menu or organize the settings menu so its just a little more intuitive whats going on for a new users.  If the menus aligned more closely to the preferences dialog it would at least help build the intuition.  I think @GregValiant distilled nicely the most confusing aspects (settings, preferences, configure all lead to the same place in different ways).  I think the preferences dialog itself its much more clear than the menus.

lorenzoromagnoli, 2022-04-15:
I tend to agree with the proposal of simplifying and merging menu entries. 
With the switch to qt6 we won't integrate the application menu on the top bar on mac as we used to do, so we have a bit more freedom in regards to how preferences entries are handled on mac. 
We could also consider removing some entries from the application menu which are very much accessible from the main Cura UI (e.g printer selection and extruders configuration)

So here are my 2 cents: 
- **Remove settings altogether.**
- **Extend preferences to have quick access to all Cura preferences:** 
  - configure cura
  - manage settings visibility
  - manage printers
  - manage materials
  - manage print profiles