Ultimaker/Cura #11779: Top Bar showing status

vikky999, 2022-04-07:
### Application Version

4.13.1

### Platform

Mac

### Printer

Creality CR-10

### Reproduction steps

Change the bed adhesion option from anything to anything. It doesnt matter what you do, the top bar still shows, bed adhesion as off even though the actual selection does work. 

### Actual results

Cura slices the model as expected, only the UI doesn't show the bed adhesion setting i choose on the top bar

### Expected results

Update the UI

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2022-04-08:
Hi @vikky999 thank you for your report.
Could you share a screenshot of your settings? At your current screenshot I can't see what you have selected.

vikky999, 2022-04-08:
I've selected Skirt as seen in the screenshot, but the bar above shows "Off"

[image]

Ghostkeeper, 2022-05-09:
This is expected behaviour. Brim and Raft are considered "on", while Skirt and None are considered "off". That is because the skirt doesn't really add to the adhesion of the print, and many printers simply use it to prime and to ensure the bed is level. Novice users shouldn't be expecting that any effort is being done to improve the adhesion.

The implementation of this is here:

https://github.com/Ultimaker/Cura/blob/6f7145bb12e3f8c03a16447bcb5d689d6f82b2f8/resources/qml/PrintSetupSelector/PrintSetupSelectorHeader.qml#L87

closed 2022-05-09 by Ghostkeeper