Ultimaker/Cura #11781: When Multiplying an object, nothing happens and the progress bar remains on screen (empty)

jasoben, 2022-04-08:
### Application Version

4.11

### Platform

OS X

### Printer

S5

### Reproduction steps

1. Edit > Multiply Objects

### Actual results

The multiply window doesn't do anything and continues to block the view even if you open another project

### Expected results

Objects should multiply

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

I'd include the project file but it won't let me upload the 3MF format

fieldOfView, 2022-04-08:
This has been fixed in later versions of Cura. Time to update!

jasoben, 2022-04-11:
Gotcha thanks!

closed 2022-06-15 by nallath