Ultimaker/Cura #11791: Ultimaker Cura Causing BlueSc

PlatinumStars, 2022-04-12:
### Application Version

4.13.1

### Platform

Windowsn 10 

### Printer

Ender 3 

### Reproduction steps

I open Cura. At any random time while cura is running in the background or while using it. The problem happens 

### Actual results

A blue Screen with the error driver_irql_not_less_or_equal comes up it sometimes happens while the computer is idle

### Expected results

Cura should not cause this problem

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8471357/cura.log)

Golf3gti1, 2022-11-05:
I have the same problem when I start Cura and after 5-10 minutes of working on the object I get a blue screen.

GregValiant, 2024-11-15:
Is this still an issue with recent versions (5.8.0 and up)?  Can I close this?

closed 2024-11-16 by GregValiant