Ultimaker/Cura #11794: Arachne Infill Speed

GregValiant, 2022-04-12:
### Application Version

Arachne Beta 2

### Platform

Windows 10 Home (onboard video OpenGL 4.0 in compatibility mode)

### Printer

Creality E3Pro

### Reproduction steps

Brought in a model
Print Speed = 50
Infill Speed = 50
Outer Wall Speed = 35
Sliced the model and saved a gcode file.

### Actual results

Within the Gcode file the infill speed starts out at 50mm/sec but after a couple of layers the infill speed drops to 25mm/sec (50%).  I looked around here and didn't find a similar report.
My expectation was that the Infill speed would be as set in Cura (in this case 50mm/sec).

This is the first instance of "Infill" in the gcode:
;TYPE:FILL
G1 **F3000** X136.98 Y100.591 E427.56652
This occurs two layers later:
;TYPE:FILL
G1 F2100 E522.23222
G1 **F1500** X142.006 Y111.653 E522.24304
Lines with speeds like this occur intermittently later in the Gcode:
;TYPE:FILL
G1 **F2027** X144.477 Y121.961 E682.40177
G1 **F2547.8** X144.773 Y122.136 E682.41571
G1 **F3428.6** X145.261 Y122.136 E682.43041
Those are three consecutive extrusions.

### Expected results

Infill speed at 100% (in this case - 3000mm/min)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Switch Box Compact.zip](https://github.com/Ultimaker/Cura/files/8474617/Switch.Box.Compact.zip)

fieldOfView, 2022-04-12:
I have not looked at the project file yet, but did you check the minimum layer time?

GregValiant, 2022-04-12:
@fieldOfView Yes.  It's a decent size model.  Outer and inner wall speeds are at 50 during a layer, and then the infill falls off to a lower number.  I really don't get the variation between extrusions.  Layer times for the lower portion of the print are between about 58 seconds when there is no infill and 75 seconds when there is infill.  Once it gets to the threads it starts bouncing off the Minimum Layer Time.

smartavionics, 2022-04-12:
Does it still do that if you set the infill layer height to 0.2 ?

GregValiant, 2022-04-12:
This is with a scaled up calibration block...
@smartavionics - It looks like the 2X layer height does cause the speed reduction.  Anticipation of the 2X flow??

It looks like the first layer of infill goes down at full speed and it is .20 high.  The following layers of infill that are .40 high go down at reduced speed.  That includes the top layer of infill which is .20 and I would have anticipated full speed for it.
So I guess this is expected behavior.  I've worked mostly with 4.13.1 lately and it does not have this feature.  I wouldn't mind being able to shut it off since for larger models it could be a serious hit to print time.  I've never had a flow problem and I run a lot of prints with 2X layer height for infill and support infill.

I'm going to leave this open for more comments.  If there is not a way to shut it off then maybe this could be a feature request.

smartavionics, 2022-04-13:
Perhaps your hitting an extrusion rate limit? Is that configurable? I don't know as I'm not familiar with the arachne build.

GregValiant, 2022-04-22:
Expected behavior.  Flow Equalization Ration was at 100% and was causing the slowdown.

closed 2022-04-22 by GregValiant