Ultimaker/Cura #11797: Fails to extrude after first layer

sgpage, 2022-04-13:
### Application Version

4.13.1

### Platform

Windows 11

### Printer

Ender 3 V2

### Reproduction steps

Carried out multiple renders of fairly basic model and all times they failed after the first 3/4 layers and then carried on printing nothing into thin air. Reverted to v4.12.1 and all worked fine.

### Actual results

Onle layer one printed. 

### Expected results

Printed model!

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3_Van Panel v3.zip](https://github.com/Ultimaker/Cura/files/8479746/CE3_Van.Panel.v3.zip)

GregValiant, 2022-04-13:
The gcode looks fine.  I don't see anything in there that would cause the printer to stop extruding.
I had some moist scrap PLA and used it to print your gcode.  I printed into layer 3 without any problem.  That was at 300% feedrate on my Ender 3 Pro.

sgpage, 2022-04-15:
Strange (may have been more layers than three thoigh as I can't easily tell how many it did) - stopped at same point 4 times. Even swapped my nozzle out thinking it was that. Old version worked like a breeze (and produced 2k smaller gcode file)

Steve Page
Tel 07703 545349

GregValiant, 2022-04-15:
You may need to take a close look at your hot end and specifically at the point where the bowden tube butts up against the nozzle.  That's a popular spot for a plug of plastic to form a blockage and all the Ender models have the problem as they share the same hot end.

If you haven't changed the extruder and still have the stock plastic one then take a look at the plastic around the extruder arm pivot hole.  All those stock arms (100%) develop a crack and fail there.  Some Enders were delivered with cracked arms.

GregValiant, 2024-11-15:
Is this still an issue with recent versions (5.8.0 and up)?  I'll remove the bug label as there is nothing wrong with the gcode and I'll go ahead and close this.

closed 2024-11-15 by GregValiant

sgpage, 2024-11-16:
No problem now thanks

Steve Page
Tel 07703 545349