Ultimaker/Cura #11799: Tree support merged with model

zifnab0, 2022-04-14:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Longer LK4 Pro

### Reproduction steps

Use tree support on this model : https://www.thingiverse.com/download:9752565

### Actual results

Tree support merged with model :
[image]

### Expected results

Tree support separated from model

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

Asterchades, 2022-04-15:
This was already fixed by 4.13.1 back at the end of January. It's the first of only two listed bug fixes that version is for.

Updating to the latest version should be a matter of course before reporting bugs; or at least reading the change logs.

closed 2022-04-15 by nallath

zifnab0, 2022-04-15:
Thank you for your answer. I apologize for posting this issue.