Ultimaker/Cura #11802: Cura 4.13.1 closes all by itself with no input

CakePiper69, 2022-04-14:
### Application Version

4.12.1 and 4.13.1

### Platform

Windows 10 Home

### Printer

Monoprice Mini V2

### Reproduction steps

Opened Cura.  Waited for about 30 seconds.
Clicked on "open file"
File explorer opened. I waited for the little blue spiral to stop rotating.
Then I clicked on the folder where the file was located. Then I waited again since I was having problems.
With no other input or mouse clicks, Cura closes.

I had this problem first with 4.13.1 so I uninstalled it and reinstalled 4.12.1. No help.  The last time I used 4.13.1, I was able to create gcode and produce a print on my Monoprice mini.  The only thing that has changed is Microsoft installed a couple of updates.

Edition	Windows 10 Home
Version	21H1
Installed on	‎6/‎27/‎2020
OS build	19043.1645
Experience	Windows Feature Experience Pack 120.2212.4170.0

Any help would be greatly appreciated.  Thank you.  Walt Beardsley   Kingston, WA

### Actual results

After selecting a folder to open a file, Cura closes all by itself with no input.

I did try opening a previously opened file, and it opened fine.  It's only when I try to open a new file that the program closes.

### Expected results

After selecting a folder, I should be able to select and open a file to print.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8492463/cura.log)

CakePiper69, 2022-04-14:
Uninstalling Chitubox seems to have solved my problem.  Not sure what the interference is, but now I can open files normally.  sigh...

closed 2022-04-14 by CakePiper69

nallath, 2022-04-15:
> Uninstalling Chitubox seems to have solved my problem. Not sure what the interference is, but now I can open files normally. sigh...

https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file 

It's caused by the preview of Chitubox. The link contains information on how to disable that so you can continue to have chitubox installed on your machine

CakePiper69, 2022-10-11:
THANK YOU SO VERY MUCH!!  🙂

https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file