Ultimaker/Cura #11803: Remembering print name

soleshoe, 2022-04-15:
### Is your feature request related to a problem?

output filename is not persistent when launching a print. The generated output file printed is stored on my MainsailOS instance so if i don't adjust filename, the stored files in mainsailOS get inconsistent.

### Describe the solution you'd like

As I am loading objects to print in Cura through Fusion360, I often use the same Cura project to print different objects at different times. For now I have to rename the output filename each time i launch a print in Cura, even if it's the same last object printed with different settings, or failed print needing restart. 

### Describe alternatives you've considered

renaming output files each time i launch a new print

### Affected users and/or printers

consistency in object filename output.

fieldOfView, 2022-04-15:
You can give the printjob a name before saving/sending it to the printer, in the lower left of the viewport window. There's a small pencil icon that you can press, but you can also click in the job name in the lower left and just start typing/editing the name that is there.

GregValiant, 2022-04-15:
#11043, #11756, #11276 are similar.  It seems that the previous open project name persists in the background (somehow).  After saving a new project - the name reverts to the previous project, and although  is reflected in the box and title bar, it is easily over-looked.
This is especially annoying when saving to SD since the user doesn't get a chance to actually see the name and so a file on the SD can be easily over-written when that was not the intent.

fieldOfView, 2022-04-18:
> This is especially annoying when saving to SD since the user doesn't get a chance to actually see the name

The jobname, which is used as a filename, is shown in the lower left corner of the viewport.

GregValiant, 2022-04-18:
@fieldOfView I understand.  What is happening is that after saving a file that "Project Name" box (and the TitleBar text) reverts back to a previous jobname without any action from the user.  If you make a change to the model and go to save it again, that earlier job name is the one used for the file rather than the jobname you just saved as 15 seconds ago.  I'll try to explain (but it is confusing).

I load "GV Vera.stl, slice it and save the gcode using the name "GV Vera".  That's all good.
I clear the build plate.  The Project Name box gets hidden and the TitleBar text is "Untitled - Ultimaker Cura"
I open "Test Nut".stl.  The Project Name is now "Test Nut" and the TitleBar text is "Test Nut - Ultimaker Cura".   I slice it and save the gcode and it saves correctly as "Test Nut.gcode".  Still all good.

But wait, I forgot to move the Z seam.  I change the setting and save the gcode to the Removeable Drive but it gets saved as "GV Vera.gcode".  That overwrites my previous gcode file on the SD, but I don't know that until I go to print the file and it's the wrong model.

Immediately after saving Test_Nut.stl as Test_Nut.gcode the project name changes from Test Nut to a previous project that was named GV Vera.  This happens every time.  I cut the model out of this image but the "Test Nut" is still loaded.
[image]

DanielFZahn, 2022-05-27:
@GregValiant has explained the problem annoying me perfectly. After clearing the build plate, I can load new models and save their gcode under the new model name, but when I go to save the project, it's the old models name. Why can it not maintain the name from the gcode save, I do not know and it drives me crazy.