Ultimaker/Cura #11805: Start over again the infill in bottom layer

djdoubleg, 2022-04-15:
### Application Version

4.13.1

### Platform

Windows10

### Printer

Self Made

### Reproduction steps

Slice this stl.
https://www.thingiverse.com/thing:4462331/files (Touch mi base)

### Actual results

Start bottom layer infill twice

### Expected results

Only one make the layer

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://user-images.githubusercontent.com/83548720/163629646-4fb7cf9c-2729-47e9-b8be-8df5f06f9334.mp4

GregValiant, 2022-04-15:
Yes is does.  I guess Monotonic ordering does not work with Zig Zag.  You can turn off Monotonic or you can change the line type to "Lines".
This behavior has been mentioned before. https://github.com/Ultimaker/Cura/discussions/10931#discussioncomment-1886436

Ghostkeeper, 2022-05-20:
I couldn't reproduce this issue, not in 5.0 nor 4.13, using Custom FFF Printer on . Has it been fixed in 5.0? Do you have a project file for us that displays the issue?

This is the project file that I tried: 
[supposed_zigzag_monotonic_repeating.zip](https://github.com/Ultimaker/Cura/files/8739135/supposed_zigzag_monotonic_repeating.zip)

closed 2022-06-03 by github-actions[bot]