Ultimaker/Cura #11806: Machine start G code issue

simonwebnet, 2022-04-15:
### Application Version

4.13.1

### Platform

MacOS

### Printer

FLsun SR

### Reproduction steps

setting first layer bed temperature different from general (profile panel in "All" mode)

### Actual results

The generated G code send in machine start G-Code: M140 S{material_bed_temperature} the general temperature but the first layer (layer 0) does not have M140 refresh so it print at general temperature and next layer (layer 1) refresh M140 at general temperature (redundant in this case).

### Expected results

The layer 0 has M140 from mine first layer temperature setting or the start G-Code: M140 S{material_bed_temperature} receive the setting that is in first layer temperature field. (second option is better so there isn't lag to reach the temperature while printing)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Hope is understandable enough.
Thanks

GregValiant, 2022-04-15:
Try:
M140 S{material_bed_temperature_layer_0}
and for the hot end:
M104 S{material_print_temperature_layer_0}
They also work with M109 and M190
@fieldOfView has put together a list of the replacement words [HERE](http://files.fieldofview.com/cura/Replacement_Patterns.html).

GregValiant, 2024-11-15:
I'll go ahead and close this as it was not a Cura issue.

closed 2024-11-15 by GregValiant