Ultimaker/Cura #11810: slicing fail

abrassfield, 2022-04-17:
### Application Version

4.13.1

### Platform

Windows 11

### Printer

svol 1

### Reproduction steps

1. used support blockers on moon knight helmet 
2. sliced and a few min in it stops and gives me a popup with slicing failed and ask me to do a bug report

### Actual results

sliced and a few min in it stops and gives me a popup with slicing failed and ask me to do a bug report

### Expected results

ready to export to print.
[cura.log](https://github.com/Ultimaker/Cura/files/8502020/cura.log)
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

it is not letting me upload project
[cura.log](https://github.com/Ultimaker/Cura/files/8502034/cura.log)

GregValiant, 2024-11-15:
Is this still an issue with recent versions (5.8.0 and up)?  Can I close this?

abrassfield, 2024-11-15:
Honestly I can't recall it's been so long ago since I placed this ticket.
And I haven't used the slicer in a long time. You can go ahead and close
the ticket and if I find out there's another issue I can reopen a new one.

closed 2024-11-15 by GregValiant