Ultimaker/Cura #11812: Right click overloaded

Fyreye, 2022-04-18:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Snapmaker 2

### Reproduction steps

navigate the 3d space, right click to orbit

### Actual results

about half the time I try to orbit it opens the right click menu and selects and executes the nearest option in the menu to my cursor causing my plate to get rearranged or models deleted

### Expected results

the object to rotate in my view so I can see different areas of it

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

having the right click be both menu and rotate/orbit makes navigation a nightmare with no way to rebind it to shift, control or alt click of any kind

GregValiant, 2022-04-18:
What mouse?

Fyreye, 2022-04-18:
an older Razer Naga

GregValiant, 2024-11-15:
Is this still an issue in recent versions (5.8.0 and up)?  Can I close this?

closed 2024-12-09 by github-actions[bot]