Ultimaker/Cura #11814: Problem creating letters with inner spaces

Pebels2626, 2022-04-18:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Artillery Sidewinder X1

### Reproduction steps

I've created a text with Blender and exported it as STL, I've done it 1000 times so I don't think I made any mistakes in the design.

### Actual results

A hole that the laminate fills is not processed.

### Expected results

The central space of the letter "e" should appear without padding.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Attached two images about the problem.
[image]
![image](https://user-images.githubusercontent.com/103932414/163794423-da55b47a-116c-46f1-afeb-6e446b951ed9.png
[ASX1_Vera.zip](https://github.com/Ultimaker/Cura/files/8505053/ASX1_Vera.zip)
)

GregValiant, 2022-04-18:
If you would use the "File | Save Project" command in Cura and then zip the 3mf project file and post it here it may give clues to what is happening.

Pebels2626, 2022-04-18:
i did it. The file is ant the end of the post, it is called "ASX1_Vera.zip"

GregValiant, 2022-04-18:
The "r" and the center of the "e" are the problems.  I was able to repair the STL in MS 3D Builder.
Prior to repairing the STL It sliced kind of as expected after I used the Mesh Tools to split the model into parts and then play with the Mesh Fixes settings in Cura.
This is the repaired model.  Zooming in I can still see a bit of funkiness between the E and R and between the R and the A.

[image]

Pebels2626, 2022-04-30:
can you share the model fixed please?

GregValiant, 2022-04-30:
You can fix it in several different free softwares.
[ASX1_fixed_Vera.zip](https://github.com/Ultimaker/Cura/files/8597466/ASX1_fixed_Vera.zip)

Pebels2626, 2022-04-30:
I finally download MS 3D builder and fixed myself, thank you so much anyway!! :3

closed 2022-04-30 by Pebels2626