Ultimaker/Cura #11815: Filament change with ender 6 does not work

Magic2226, 2022-04-19:
### Application Version

4.13.1

### Platform

Windows 11

### Printer

Creality Ender 6

### Reproduction steps

1. Extensions, post processing, modify G Code
2. Added 3 scripts to pause and place nozzle at x20 y20 on each of them.
3. Passed the 1st one and second one paused and moved to x20 y20
4. I replaced filament and it then tried to restart print from beginning.
5. NEXT
6. I tried Filament change scripts. 
7. Added 3 scripts.
8. did not stop at any layer height
9. at each script I set to move Z to 100mm
10. Did not stop once so I had to cancel

### Actual results

With PAUSE, it skipped 1st one and on second script it moved to where I told it but would not have any option to do anything but restart then tore into the 1st part of print.

### Expected results

Expected it to stop at layer 18 and move to y20 x20 so I could change filament. Did not happen.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Have tried 4 times and did not work.

GregValiant, 2022-04-19:
Slice a calibration cube and pick a layer for a Pause at Height and another for a Filament Change.  Slice and save the gcode.
Put the gcode into a zip folder and post it here.

When Creality moved to the 4.2.x mainboards there was poor integration with the TFT type LCD displays.  Commands that send a message to the LCD (M0, M1, M117, M600, etc.) no longer worked.  That may not be your problem, but it's the most likely cause.  The commands you are talking about work fine on my older Ender 3 Pro with the 1.1.5 mainboard.

Magic2226, 2022-04-19:
Will do this when I get off work. I did get a red message something about m0 being a bad command.Sent via the Samsung Galaxy Note20 5G, an AT&T 5G smartphone

  (Issue #11815)

Magic2226, 2022-04-19:
I have picked three.  

Pause at height layer 20Filament change at layer 40Filament change using printer firmware at layer 80

GregValiant, 2024-11-15:
I'll go ahead and close this.  It is likely that M600 was not enable in your firmware.  I hope you understand.

closed 2024-11-15 by GregValiant

Magic2226, 2024-11-15:
I had a problem like this too. does not seem to work running through the pi but from thumb drive mine works.Sent via the Samsung Galaxy S24 Ultra, an AT&T 5G smartphone

  (Issue #11815)