Ultimaker/Cura #11834: Unexpected, unstyled scrollbars in tooltips

fieldOfView, 2022-04-21:
### Application Version

5.0

### Platform

Windows 10

### Printer

all

### Reproduction steps

1. In the "Custom" settings bar, view the tooltip for eg "Layer Height" or "Line Width"
2. Mouse over to the tooltip and scroll the (unexpected) horizontal tooltip

### Actual results

[image]

### Expected results

No horizontal scrollbar, and a styled vertical scrollbar (only if necessary, which it should not be in the screenshot; there is room for the whole text in the viewport).

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional info.

ungutknut, 2022-04-21:
I can verify this bug on Win10. Additionally this scrollbar sometimes just doesn't appear and text is just cut-off. If it appears it's useless because the tooltip text disappears before I can hover the mouse on it. On Linux (Xubuntu) the tooltip is also cut off with no scrollbars at all.
[image]

fvrmr, 2022-04-25:
Hi @fieldOfView thank you for your report.
We are aware of this issue and have a ticket on our backlog for this. Devs see CURA-9112

@ungutknut I will add to the ticket that the tooltip is cut off.

redstone99, 2022-04-25:
A screenshot of clipped text in tooltip
[image]