Ultimaker/Cura #11843: Lines in Vasemode

dyoga, 2022-04-21:
### Application Version

5.0 beta 1

### Platform

Windows 11

### Printer

Creality CR-6 SE

### Reproduction steps

Sliced a model with the option Vasemode

### Actual results

1. Strange Lines at Vasemode sliced and get lines between walls

### Expected results

No lines in the middle (betwwen walls)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CCR6SE_Pencil_holder.zip](https://github.com/Ultimaker/Cura/files/8531984/CCR6SE_Pencil_holder.zip)
[image]

fvrmr, 2022-04-25:
Hi @dyoga thank you for your report.
We are aware of this issue and have a ticket on our backlog for this. Devs see CURA-7925

fvrmr, 2022-04-25:
As a workaround you can enable `Remove all holes` which will remove those lines. But don't forget to disable this if you don't use Spiralize!

theolas, 2022-09-15:
I have this issue too
[image]
https://www.thingiverse.com/thing:4660786/files Vase_01_SMALL.stl