Ultimaker/Cura #11844: all my Plugins had to be disabled/unistalled and no plugins in marketplace

jaymcd, 2022-04-21:
### Application Version

5.0 Beta 1

### Platform

Windows 11

### Printer

Adimlab Gantry

### Reproduction steps

Happens each time I open the app

### Actual results

No plugins showing in Marketplace. I installed 5 Beta 1 -0 still have 4.13 installed. It pulled plugins config from 4.13

### Expected results

Available plugins should be there or previous should work/stay installed.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

NA

lukas5450, 2022-04-21:
have the same issue on 5.0.0-beta+1 and after disabling the plugins and restarting Cura is becomes stuck starting "initializing package manager..."

nallath, 2022-04-22:
Duplicate of https://github.com/Ultimaker/Cura/issues/11858

closed 2022-04-22 by nallath