Ultimaker/Cura #11845: OpenGL Error on Linux AMD Systems with Mesa

NerdHarder615, 2022-04-21:
### Application Version

5.0

### Platform

Linux

### Printer

Creality Ender 5 Plus

### Reproduction steps

Downloaded Ultimaker-Cura-5.0.0-beta+1-linux.AppImage and tried to open the application, received and error about OpenGL missing

### Actual results

Can't open or run the appimage

### Expected results

Application opens and runs correctly

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I am not able to run the AppImage on Arch Linux. I get the following error message:
Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers.

No log files have been generated in ~/.local/share/cura/5.0

I am running the current version of Mesa on my system. (AMD CPU & GPU)
> sudo pacman -Q |grep -i mesa
lib32-mesa 22.0.1-3
libva-mesa-driver 22.0.1-3
mesa 22.0.1-3
mesa-utils 8.4.0-7

> glxinfo |grep OpenGL
OpenGL vendor string: AMD
OpenGL renderer string: AMD Radeon RX 6700 XT (navy_flounder, LLVM 13.0.1, DRM 3.44, 5.17.4-zen1-1-zen)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 22.0.1
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6 (Compatibility Profile) Mesa 22.0.1
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 22.0.1
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

jellespijker, 2022-04-21:
Please add the log files as well.

nallath, 2022-04-22:
Duplicate of https://github.com/Ultimaker/Cura/issues/11856 

There is a possible fix in that ticket.

closed 2022-04-22 by nallath