Ultimaker/Cura #11847: Flow Equalization ratio not changing the actual print speed

fakcior, 2022-04-21:
### Application Version

5.0.0 beta+1

### Platform

Windows 10

### Printer

Artillery Genius

### Reproduction steps

1. Set outer wall speed to 70mm/s
2. Set outer wall flow to 120%
3. Flow equalization ratio left at 100%

### Actual results

Outer wall speed still at 70mm/s

### Expected results

Outer wall speed should be about 58mm/s

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

https://file.io/BQRQtF4IpZCG

fvrmr, 2022-04-26:
Hi @fakcior thank you for your report.

Changing the Flow settings does not have an effect on printing speed. Only if the flow rate is changed when having variable line width in thin parts on a model. So in this case this it is expected that the out wall speed is 70mm/s
The Flow Equalization ratio only applies to the width changes introduced by the variable line width. With your model there are no variable line widths. 

Let me know if this answer your issue and reach out if you have more questions.

fakcior, 2022-04-26:
Thank you, I didn't know that is the expected behaviour since in Cura Arachne Engine Beta 2, which I was using before, it certainly was changing the speed based on flow rate.

closed 2022-04-26 by fakcior