Ultimaker/Cura #11848: (~4.10+) Corrupted rendering of lines in preview

Haxk20, 2022-04-21:
### Application Version

I think it started around 4.10 and all other versions after that.

### Platform

Linux Gentoo, Radeon RX5600XT

### Printer

VORON V2.4, Tevo Tarantula Pro

### Reproduction steps

When trying to preview the gcode of the model the layers below the currently selected one are shown as corrupted.
This started few months ago and I cant seem to figure out what may be causing this bug. I tried checking rendering for the GPU driver (Mesa Open source stack) but all other software renders correctly so the issue is specific only to cura.

As for reproducing. Load any model and look at it in preview mode.

### Actual results

The result is this:

In this screenshot you can see the rendering working correctly as only a single layer is selected.
[image]

Here rendering goes corrupt on all other then currently selected layer.
[image]

Same here.
[image]

### Expected results

Expected would be correct rendering of all layers.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Attaching a cura log. Not including project file as this happens with any file you slice.
[cura.log](https://github.com/Ultimaker/Cura/files/8534163/cura.log)

While only a cosmetic issue its VERY hard to sometimes see what may be happening in the gcode file and its hard to judge
issues that may show up in gcode.

closed 2022-04-21 by Haxk20

tapir, 2022-04-22:
I have the same issue. May I ask why is this closed?

Haxk20, 2022-04-23:
> I have the same issue. May I ask why is this closed?

As its a mesa rendering bug it seems. Does also occur on old versions of cura so it cant be cura bug