Ultimaker/Cura #11850: "Use single Instance of Cura" causes crashing on startup.

Divenity, 2022-04-21:
### Application Version

5.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Start Cura
Enable "Use single Instance of Cura" in preferences
Restart Cura

### Actual results

Cura crashes upon startup. with a crash report stating the following:

"**OopsUltimaker Cura has encountered something that doesn't seem right.**

We encountered an unrecoverable error during startup. It was possibly caused by some incorrect coniguration files. We suggest to backup and reset your configuration

Backups can be found in the configuration folder.

Please send us this Crash Report to fix the problem."

### Expected results

Cura starts as normal.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Error traceback:
Traceback (most recent call last):
  File "cura_app.py", line 238, in <module>
  File "cura\CuraApplication.py", line 813, in run
    super().run()
  File "UM\Qt\QtApplication.py", line 366, in run
    super().run()
  File "UM\Application.py", line 256, in run
    self.initialize()
  File "cura\CuraApplication.py", line 323, in initialize
    self.__sendCommandToSingleInstance()
  File "cura\CuraApplication.py", line 339, in __sendCommandToSingleInstance
    if self._single_instance.startClient():
  File "cura\SingleInstance.py", line 32, in startClient
    if single_instance_socket.state() != QLocalSocket.ConnectedState:
AttributeError: type object 'QLocalSocket' has no attribute 'ConnectedState'

Log:
[cura.log](https://github.com/Ultimaker/Cura/files/8534364/cura.log)

Only way to get Cura tp start again is to reset configuration, losing all printer and print settings.

Vandresc, 2022-04-22:
Thank you @Divenity , devs see CURA-9183 for our own internal reference.

nallath, 2022-04-22:
https://github.com/Ultimaker/Cura/commit/c1ed7b97f9e5083c2571d60d2e1a031ee48fd9ca

pilot76103, 2022-04-22:
I can confirm this issue happens on macOS as well.

Vandresc, 2022-04-22:
Fixed! Thank you all for your feedback.

closed 2022-04-22 by Vandresc