Ultimaker/Cura #11851: Error Segmentación

Blankita1999, 2022-04-21:
### Application Version

5.0.0 beta

### Platform

win 10

### Printer

Geeetech A10

### Reproduction steps

Metí un STL de una luna, y al querer segmentarlame da error, no porque tenga puesto nada mal, simplemente me pone "Se produjo un error, haga click aqui para informar de este error" o algo así.
Luego también está dando problemas con las letras de las descripciones, que se juntan unas con otras.
Y a la hora de meter un archivo STL 90MB tarda muchisimo, en la versión de 4.13.0 (si no me equivoco de version) nada de esto pasa con ese mismo stl.
Envío esto para ayudaros a corregir problemas.
Un saludo.
Gracias

### Actual results

[image]

### Expected results

nada avanza

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nome deja poneros archivos stl

jellespijker, 2022-04-21:
Please upload log file and project 3mf. Otherwise it is impossible for us to analyze the issue.

fvrmr, 2022-04-25:
You can find your log file here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`
Please share your files in a .zip here on GitHub.

closed 2022-05-23 by github-actions[bot]