Ultimaker/Cura #11853: Only one plugin will show on the marketplace.

N0edL, 2022-04-21:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

I Went to the marketplace and only one plugin is there.

### Actual results

I clicked on the marketplace and only **Barbarian Units** is there.

### Expected results

Does Someone else have this or a solution for it.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

N0edL, 2022-04-21:
someone?

jellespijker, 2022-04-21:
This Cura version has a major SDK increase. Plugin maintainers have to change and fix their code to ensure their plugin is compatibel with Cura 5.0.

This is because we upgraded from PyQt5 to PyQt6 for our graphical interface

nallath, 2022-04-22:
See also https://github.com/Ultimaker/Cura/issues/11858

fvrmr, 2022-04-25:
Closing this as a duplicate of #11858

closed 2022-04-25 by fvrmr