Ultimaker/Cura #11856: Cura 5.0 beta wont launch on Ubuntu 2022.04

leoheck, 2022-04-21:
### Application Version

5

### Platform

Ubuntu

### Printer

None

### Reproduction steps

Install it and launch it

### Actual results

```
➜  Applications ./Ultimaker-Cura-5.0.0-beta+1-linux_c714ec65293e8f948ace6fd69440b1d4.AppImage 
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_UltimaXxmREJ/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libLLVM-13.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_UltimaXxmREJ/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libLLVM-13.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_UltimaXxmREJ/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libLLVM-13.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
^Z
[1]  + 768770 suspended  ./Ultimaker-Cura-5.0.0-beta+1-linux_c714ec65293e8f948ace6fd69440b1d4.AppImage
```

### Expected results

See the Cura window.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Using
```
Ultimaker-Cura-5.0.0-beta+1-linux_c714ec65293e8f948ace6fd69440b1d4.AppImage
```

[image]

```
➜  Applications lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04 LTS
Release:	22.04
Codename:	jammy
```

leoheck, 2022-04-21:
Also, there is no new icon, there is no icon, nothing.
It is the second icon here.
[image]

leoheck, 2022-04-21:
Ah, Cura 4.13.1 is working fine.

nallath, 2022-04-22:
Could you try if LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage

Fixes it? We already had one report that this would resolve it

NerdHarder615, 2022-04-22:
I just tried the LD_PRELOAD and received this error:
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage
ERROR: ld.so: object '/usr/lib/x86_64-linux-gnu/libstdc++.so.6' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/usr/lib/x86_64-linux-gnu/libstdc++.so.6' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/usr/lib/x86_64-linux-gnu/libstdc++.so.6' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/usr/lib/x86_64-linux-gnu/libstdc++.so.6' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/usr/lib/x86_64-linux-gnu/libstdc++.so.6' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
qt.qpa.wayland: EGL not available

oliof, 2022-04-22:
This is an issue with library incompatibilities between the graphics drivers and libstdc++ that comes with the AppImage. FreeCAD AppImage (any AppImage with applications requiring shared library loading between the App and your X11/Wayland environment) has the same issue.

As Ubuntu 22.04 was released today only, it's only partially surprising that it is not fully supported IMO. 

The LD_PRELOAD workaround above worked for me on pop!_os 22.04 which is based on Ubuntu 22.04 (and also still in beta). You may need to find the right path to the libstdc++ your graphics drivers use to make it work.

This issue is inherent with AppImage. According to the AppImage best practice you should package the standard libs that cover the oldest OS release you want to support. That this breaks shared library loading for people that use current graphics hardware and driver libraries seems of little import to them.

From my point of view, Ultimaker has two options,  neither of them compelling: 

1. Build Cura on the now released Ubuntu 22.04. This may or may not break the build on older releases and requires investigation how far back a build is supported.

2. Provide multiple builds which will lead to confusion when people try to run the wrong build.

(and to cut to the chase, neither flatpack nor snap are free from issues either, so switching the packager is not a solution IMO).

leoheck, 2022-04-22:
@nallath the following process does not solve for me.
```
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-beta+1-linux_c714ec65293e8f948ace6fd69440b1d4.AppImage
```

leoheck, 2022-04-22:
@oliof cool, these are the libs I could find. None of them are working with Cura 5.0, however, Cura 4.13 does work without any special setup. Maybe something else is missing in this AppImage. And yes, I saw the same issue with the Appimage of FreeCad

```
➜  Applications sudo find /usr -name "libstdc++*"   
/usr/lib/x86_64-linux-gnu/libstdc++.so.6
/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30
/usr/lib/gcc/x86_64-linux-gnu/11/libstdc++.so
/usr/lib/gcc/x86_64-linux-gnu/11/libstdc++.a
/usr/lib/gcc/x86_64-linux-gnu/11/libstdc++fs.a
```

leoheck, 2022-04-22:
Ah, it is related with the graphics card. Alright. I will have to check how it works because I am using NVIDIA drivers.

leoheck, 2022-04-22:
Weird that my lib has a reference to this `GLIBCXX_3.4.30`

```
➜  Applications strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
GLIBCXX_3.4.22
GLIBCXX_3.4.23
GLIBCXX_3.4.24
GLIBCXX_3.4.25
GLIBCXX_3.4.26
GLIBCXX_3.4.27
GLIBCXX_3.4.28
GLIBCXX_3.4.29
GLIBCXX_3.4.30
GLIBCXX_DEBUG_MESSAGE_LENGTH
```

But it looks like when I run Cura, it uses a lib from inside the AppImage.
```
➜  Applications LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-beta+1-linux_c714ec65293e8f948ace6fd69440b1d4.AppImage
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_Ultimap8B3jp/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libLLVM-13.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_Ultimap8B3jp/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libLLVM-13.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_Ultimap8B3jp/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libLLVM-13.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
```

oliof, 2022-04-22:
Yes, that is how AppImage works, it collects the runtime and standard libraries from the system it is built on. and GLIBCXX_3.4.30 is not covered in the one that's included here because it wasn't part of Ubuntu releases prior to 22.04 which wasn't a stable release until today.

No idea why the LD_PRELOAD workaround does not work on your machine, since I have the same layout. Here is the wrapper script I use to reference in my .desktop file (but it also can be used on the command line of course):

```bash 
#!/bin/bash
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 <path_to>/AppImages/Ultimaker-Cura-5.0.0-beta+1-linux.AppImage 
```
I am using an absolute path, but I have confirmed that it also works when I use a relative path. Since your shell prompt looks slightly non-default in your examples, maybe you are not using bash? 

running it yields the following output, and Cura 5 starts up fine:

```bash
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltimaLZurP1/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimaLZurP1/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
```

oliof, 2022-04-22:
FWIW I went through building an AppImage using [Cura Build Environment](https://github.com/Ultimaker/cura-build-environment) on pop!_os 22.04 and as expected that launches without error. I do not have access to older environments to check whether this would  run on 20.04 / 18.04 releases.

It misses some built-in plugins, but that's more likely due to my lack of experience with the build system than anything else.

leoheck, 2022-04-22:
could you share your appimage so I can test it on my system @oliof ?

leoheck, 2022-04-25:
I was able to run it. It looks like preloading the system lib with `LD_LIBRARY` is doing nothing on my system.

This is what I've done to run it once.
```
# Extracting the package
./Ultimaker-Cura-5.0.0-beta+1-linux_c714ec65293e8f948ace6fd69440b1d4.AppImage --appimage-extrac

# Going inside the extracted package
cd squashfs-root

# Remove incompatible lib that is not allowing system lib work
mv libstdc++.so.6 libstdc++.so.6.removed.by.me

# Then I could launch it with ./Ultimaker-Cura
```

oliof, 2022-04-25:
you need to use LD_PRELOAD, not LD_LIBRARY. But I've had a discussion with the AppImage Core developer on Twitter and they've pointed nallath towards a solution that will allow dynamic loading of of system libstdc++ if needed.

-------- Original-Nachricht --------

fvrmr, 2022-04-26:
Thank you all for reporting!
I created a ticket for this, devs see CURA-9205

leoheck, 2022-05-02:
Extra things I saw in this release
- Desktop icon is missing.
- Sign in to the Ultimaker platform is not working
- Octoprint plugin is not working

oliof, 2022-05-02:
the author of the Octoprint plugin has released a prerelease version that provides basic functionality (connectivity and upload of files), but requires more work. This is not surprising as all plugins need updating for Cura 5 due to the library and API changes.

The icon works in the task switcher, I don't see it anywhere else but I start Cura through a script so I don't know how that'd work

leoheck, 2022-05-02:
Cura does not load .stl files too. I think it is not usable on Linux yet.

oliof, 2022-05-03:
it does open STL files fine for me. Maybe its best to open another issue for this topic?

-------- Original-Nachricht --------

leoheck, 2022-05-03:
I would love to open another issue @oliof , but there are too many fields and required things. Too much text I have to read.. seriously, this template to report issues, sucks all the energy I have to create the bug reports. I have already started a new issue here at least 4 times this week. But I dropped the idea after some time since I have to fill all those required things. 

It is crazy that these things are getting horrible with time and are a show stopper for reporters like me. I do these things while I am working. So, it has to be quick and easy. And it can be improved later. This repo could at least have an alternative blank template to report issues, seriously.

nallath, 2022-05-03:
> It is crazy that these things are getting horrible with time and are a show stopper for reporters like me. I do these things while I am working. So, it has to be quick and easy. And it can be improved later. This repo could at least have an alternative blank template to report issues, seriously.

You can use a blank one, but we would rather not have people use that. The amount of poor reports that we get without it is simply staggering. We also have to do this during work, so we want to make sure that we can do this somewhat efficient. I understand that it's a balancing act for both sides, but filling in the template takes less time than it does to write the post you just did ;)

oliof, 2022-05-03:
Personally as an end user I think you're just lacking the courtesy to your fellow community members trying to follow specific issues that are being tracked by word salading everything in here. Contrary to you I made an effort to understand the issue and helped find a fix for future releases, and I'm only interested to read about progress on this issue here.

So besides the fact that you're incredibly rude to the developers on this project, you're also making it hard for me to follow this specific issue by insisting to treat this issue as if you owned it. You don't.

-------- Original-Nachricht --------

leoheck, 2022-05-03:
> but filling in the template takes less time than it does to write the post you just did ;)

@nallath  Well, this is your opinion. Not a fact. But that is fine.

> Personally as an end-user I think you're just lacking the courtesy to your fellow community members 

I don't think you understand the point here. By the way, are you saying that you don't want to have users' feedback on issues and features? Because this is what I am getting here. Seriously. If you are saying that, those 1.6k issues that help to improve the platform are useless, right? Is this what you are saying? 

What is the problem with having a way to submit a blank issue? This gives flexibility to users like me to want to contribute but don't have time to spend filling out useless forms. 

I am not even asking you to fix these issues. I am just returning what I am seeing on the screen, so you can have your product improved. I am not asking or demanding anything. I am using my free time, to help you guys, to be aware of issues there your software is going to find when released. You should not treat users like that. Seriously. 

But moving one here... why don't we have a blank template for reporting issues instead of jailing all users, who want to contribute to the project, forcing them to fill a poor form?

_Gesendet vom Gmail Desktop_

oliof, 2022-05-03:
@leoheck I am not saying what you claim and your bad faith argument is not buying you any points.

As @nallath said, you _can_ submit a blank template issue, but the developers _ask_ you to use the template. You could consider being corteous, or ignoring the request and not use the template, or you can continue poisoning this issue. Either path reflects mostly on you.

Also, if you think you earn any points by insulting me although I am just a user of the product -- like you, only with some empathy for the developers -- I think you are mistaken. Seriously (-: 

PS: To make it clear: I have no affiliation to Ultimaker or Cura other than being an end user. And I would like to follow this issue to see whether the library dependency issue on distributions derived from Ubuntu 22.04 will be resolved in a future _beta release_ or not. I would like to follow other issues in, well, other issues.

piontec, 2022-05-04:
FYI: I'm on wayland on ubuntu 22.04 and I can start cura 5.0 beta AppImage with
```
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 cura-beta
```

but it fails when I configure for wayland:
```
➜  QT_QPA_PLATFORM=wayland LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 cura-beta
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
/tmp/.mount_cura-bfnDvPl/Ultimaker-Cura: symbol lookup error: /lib/x86_64-linux-gnu/libEGL_mesa.so.0: undefined symbol: wl_proxy_marshal_flags
```

Ghostkeeper, 2022-05-04:
I'm using Ubuntu 22.04, with a fairly standard set-up, just the default interface and such. The Cura 5.0 beta AppImage starts fine for me. No issues, except from that it slows down the dragging of windows across all of GNOME (low framerate) which has now been fixed for stable (thanks to Nallath). I'm currently the only Cura developer that has a development environment on Ubuntu 22.04 with which we could debug, but it's unlucky that it works correctly for me then.

I have Nvidia drivers 470.103.01 installed, in case that's relevant:

    nvidia-driver-470/jammy,now 470.103.01-0ubuntu2 amd64 [installed]
    libstdc++6/jammy,now 12-20220319-1ubuntu1 amd64 [installed,automatic]

There was a [Twitter response](https://twitter.com/probonopd/status/1518465343953547265) from Probono that supposedly bundles libstdc++ and then checks at run-time whether it is supposed to use it (also referred to [earlier in this thread](https://github.com/Ultimaker/Cura/issues/11856#issuecomment-1108930626)). I think that refers to [AppImage-builder](https://github.com/AppImageCrafters/appimage-builder). I'm not sure if we can make it for the 5.0.0 release to transition to that, but we'll have to see.

dsvi, 2022-05-10:
For me it fails on the 22.04 with:
`/tmp/.mount_UltimaLRlMcj/Ultimaker-Cura: symbol lookup error: /lib/x86_64-linux-gnu/libEGL_mesa.so.0: undefined symbol: wl_proxy_marshal_flags`

4.13 starts fine though.

Dorro101, 2022-05-13:
On new install of Fedora 36, using Wayland, Gnome 42, had the same issue. Tried `LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 cura-beta`, which did not work. On closer inspection, the path `/usr/lib/x86_64-linux-gnu/` and has been replaced by `/usr/lib64`, so I used `LD_PRELOAD=/usr/lib64/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage` instead. Seems to be working now...
[image]
[image]
Busy with other post-install stuff, clean install to get rid all the damage my random fiddling causes. I guess some judicious symlinking would fix this, will try when done re-fiddling....

probonopd, 2022-05-13:
As pointed out by @Ghostkeeper, from what I understand https://github.com/AppImageCrafters/appimage-builder has a solution for this. cc'ing @azubieta who is the expert here.

If I am not mistaken, it roughly follows an approach roughly similar to https://github.com/darealshinji/AppImageKit-checkrt/.

azubieta, 2022-05-13:
@probonopd libstdc++ selection at startup is still in our TODO list. Only libc swtiching is supported right now.

dmitrygribenchuk, 2022-05-16:
> Could you try if LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage
> 
> Fixes it? We already had one report that this would resolve it. It's not a proper fix, but it might work as a workaround

Hi. A have the same issue on Ubuntu 22.04 (Intel Graphic Card).
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage - working fine

LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Cyclic dependency detected between "file:///tmp/.mount_UltimahwauC4/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimahwauC4/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext

without:
./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage QT_QPA_PLATFORM=wayland 
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
libGL error: MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast

Have the UI bug with a printer list but it's not related this topic
[image]

Dorro101, 2022-05-18:
Same issue with the new Cura 5.0.0 final, managed to solve the issue using `LD_PRELOAD=/usr/lib64/libstdc++.so.6 /home/dorian/.local/share/appimages/Ultimaker-Cura-5.0.0-linux.AppImage`, I had to change the path from `/usr/lib/x86_64-linux-gnu` though.

oliof, 2022-05-18:
flatpack has its own set of issues, judging other packages. I don't think tool hopping is the right approach here. Cura devs got pointed toward solutions which they will probably implement in the future.

dsvi, 2022-05-18:
> Cura devs got pointed toward solutions which they will probably implement in the future.

Do you mean `LD_PRELOAD=/usr/lib64/libstdc++.so.6` ? That alone nogonna work. They will simply hit another problem after that. And then even more under wayland. Flatpak at least has QT built for you with all the required QPA. So basic runtime will be the same for all the platforms. And if it works for Fedora it's gonna work for Ubuntu as well. You won't have to deal with the `.so` incompatibility mess.

>flatpack has its own set of issues, judging other packages. 

You should give it another try. It progresses fast.

oliof, 2022-05-18:
no the LD_PRELOAD is just a workaround, not a solution. And one could argue the same about AppImage as you do about flatpack. AppImage devs are aware of this issue and offered help here.

That said, noone is stopping you from providing flatpack builds (-:

dsvi, 2022-05-18:
I've asked cura devs about their reasoning on flatpak vs appimage. If they had any at all.
I don't remember asking you anything.

oliof, 2022-05-18:
I don't owe you receipts about my communication with the Cura devs.

dsvi, 2022-05-18:
I don't ask anything from you pal. Neither your receipts, nor your opinions.
Keep it all for yourself. Or at least try to.

nallath, 2022-05-18:
> I've asked cura devs about their reasoning on flatpak vs appimage. If they had any at all. I don't remember asking you anything.

Calm down. It's a public issue tracker, so people are free to weigh in. If you don't want that, I don't think that this is the right medium for you to have this conversation. What Oilof is sharing is also correct and good addition to the discussion. 

As for the choice between appimage and flatpack, both have their own advantages and disadvantages and we chose appimage way back when. Since change is expensive, we never switched.

Dorro101, 2022-05-18:
Flathub still on 4.13.1, will try when 5.0.0 final lands, not super worried as I have a work around, but would appreciate a fix for the current download for 5.0.0 final. I haven't used 5.0.0 much, so have yet to run into any other issues. Peace out!

dsvi, 2022-05-18:
> What Oilof is sharing is also correct and good addition to the discussion.

The only thing Oilof shared on the subject is: 

> flatpack has its own set of issues, judging other packages.

Which is hardly a "good addition". Rather a vague useless rubbish.

> Since change is expensive, we never switched.

This, on the other hand, is a technical argument. So you are using AppImage not because "flatpack has its own set of issues" but because switching at the point is too costly.
Got it. Thanks.

nallath, 2022-05-18:
> Which is hardly a "good addition". Rather a vague useless rubbish.

You're free to disagree. But if you're going to cast that stone, don't make comments like `They will simply hit another problem after that.` either. That is even _more_ vague (and thus by your own standards, even more useless rubbish). Oilof has contributed in the past and has helped out quite a bit on this issue. 

If you want to argue that someone's statement is wrong, do that instead of telling people to go away or to shut up.

dsvi, 2022-05-18:
> don't make comments like `They will simply hit another problem after that.` either.

Try to read all the messages, to have some context of the conversation. Not just rush in to save your Oliof's ego
Here is a citation for you:

>For me it fails on the 22.04 with:
/tmp/.mount_UltimaLRlMcj/Ultimaker-Cura: symbol lookup error: /lib/x86_64-linux-gnu/libEGL_mesa.so.0: undefined symbol: wl_proxy_marshal_flags

>If you want to argue that someone's statement is wrong, do that instead of telling people to go away or to shut up.

Vague rubbish has no substance to argue about. That's why it leaves no room for a constructive conversation.
If you have nothing to say in particular, just pass by. Don't use a technical discussion as an arena to show off your psycological problems or as a releif from your lonleyness (caused by the aforementioned problems).

You gotta work on your soft skills guys.
Peace out!

nallath, 2022-05-18:
> Try to read all the messages, to have some context of the conversation. 

I did. I don't see how that changes anything though. 

> Vague rubbish has no substance to argue about. 

Agreed, that is also why I'm calling out what you yourself are saying "There will be other issues" is vague rubbish that you can't argue. Either point out what those issues are, or don't make those statements. Frankly, you're being a big hypocrite here. 

> Don't use a technical discussion as an arena to show off your psycological problems or as a releif from your lonleyness (caused by the aforementioned problems).

I think you should follow your own advice and work on whatever issues you have.

dsvi, 2022-05-18:
> I did. I don't see how that changes anything though.

Not sure if you are in denial, or just have a poor reading comprehension skills. But ok. I'll give it another go:
What do you see [here](https://github.com/Ultimaker/Cura/issues/11856#issuecomment-1117025272)?
Or [here](https://github.com/Ultimaker/Cura/issues/11856#issuecomment-1122290168)?

Dorro101, 2022-05-18:
Since moving to Fedora, I prefer using flatpaks, with the intent to keep all data, system access and updates local. I will install the latest Cura as a flatpak, when available and it works. It makes sense though, to use the installation method provided by the creator, as this is the most used and tested. That said I needed to install VS Code and OpenSCAD, from the Fedora RPM repositories. The VS code flatpak lacks all the necessary system hooks and the QT on Gnome/Wayland still fuxes with OpenSCADs GUI but at least works most of the time. I had a look at AppImageKit-checkrt, it looks dated, its not clear how to apply the patch and seems more complex than my current workaround and will suffice until either a fixed AppImage (nudge, nudge, wink, wink) or a current flathub install is available (say no more, say no more)...

Ciao
Dorian

probonopd, 2022-05-21:
AppImage and Flatpak are very different. For example, Ultimaker wants you to download updates from their website, https://ultimaker.com/de/software/ultimaker-cura. Also, they want to have exact control over which Qt version is used. This is exactly what the AppImage format was designed for: To give application authors and end users 100% control over the whole experience, with no intermediaries like repositories, app stores, and the like. As a result, most AppImages are made (and supported) by the developers of the applications, whereas many Flatpaks are put together by someone "random", using "random" (usually the latest) versions of things like Qt. Which may work but may not work optimally, and is certainly not tested nor supported by the application authors.

Dorro101, 2022-05-21:
What I am alluding to, is why has this bug not been fixed in AppImage Builder yet, as it seems simple enough. The Cura 5 AppImage does not work for me out of the box and I have to do the job of adding desktop integration myself. Luckily I have the skills to fix this. I am of an age where I want one thing to do one job well and sympathise with those users who do not have the basic skills to deal with the eccentricities of Linux. I would like do this all from my software centre, without intervention. AppImage discovery and system integration into the software centre is not anywhere near the level of rpm, snap (single vendor, not cool) or flatpak (not perfect, has problems), but have no issues with any of these distribution formats, as long as they work. I would also like to see standards to allow these services to be decentralised, allowing creators to handle this themselves, if they so wish.

probonopd, 2022-05-21:
> why has this bug not been fixed in AppImage Builder yet

As @azubieta said,

> There is already a fix for it https://github.com/darealshinji/AppImageKit-checkrt/ and in the near future (provably this month) it will be ported to appimagebuilder.

probonopd, 2022-05-21:
> I would like do this all from my software centre, without intervention. 

Ultimaker wants to you download Cura from their website, though, and I can totally understand why. They are putting in lots and lots of effort and money into this software only for non-paying third-party 3d printer owners to also use it. We should consider it a great gift to the community and be thankful for it. Going to the Ultimaker website from time to time in return is more than reasonable if you ask me.

Miosame, 2022-05-21:
Besides the entitled self-assembled everyone-representing flatpak "crowd", there's shockingly those of us who prefer appimages, for exactly the reasons above and do not want anything flatpak present on our systems.

I've subscribed to this issue in case someone has a fix for arch-based systems, since passing:
`LD_PRELOAD=/usr/lib/libstdc++.so.6`
does not make it work and there is no presence of another lib on any of my systems.

The errors it throws:

```
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
libGL error: MESA-LOADER: failed to open radeonsi: /tmp/.mount_UltimaeZVcUQ/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/radeonsi_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open radeonsi: /tmp/.mount_UltimaeZVcUQ/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/radeonsi_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_UltimaeZVcUQ/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
```

Is that caused by an incompatible GPU? (Radeon HD 7790) or does it stem from libstdc still?

probonopd, 2022-05-21:
As @azubieta said,

> There is already a fix for it https://github.com/darealshinji/AppImageKit-checkrt/ and in the near future (provably this month) it will be ported to appimagebuilder.

lotw69, 2022-05-25:
Flatpak does the same as AppImage.  They just want to see who is downloading it.  If they can control what is in the AppImage so good, then people on the newest Distros that have updated libraries should not have issues.  AppImages have no protection for the user, so anyone can put bad things in the image and do nasty things.

probonopd, 2022-05-25:
> Flatpak does the same as AppImage. 

No.

nallath, 2022-05-25:
> They just want to see who is downloading it

No we don't. We can't store (or even track) that data because of GDPR. We can only track number of downloads, but that's it really.

Cyb1as, 2022-05-29:
Maybe interesting for others here, too:
I had the same problem in Opensuse Tumbleweed, with 5.0 Beta, but also with the final 5.0 release.
Preloading fixed it for me, too. Thank you very much for this hint. Library path is somewhat different in Suse, the following  line worked for me:

LD_PRELOAD=/lib64/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-linux.AppImage

lotw69, 2022-05-29:
> Maybe interesting for others here, too: I had the same problem in Opensuse Tumbleweed, with 5.0 Beta, but also with the final 5.0 release. Preloading fixed it for me, too. Thank you very much for [...]

That method there works on fedora too.  Thanks for that.

noahbliss, 2022-05-31:
> Maybe interesting for others here, too: I had the same problem in Opensuse Tumbleweed, with 5.0 Beta, but also with the final 5.0 release. Preloading fixed it for me, too. Thank you very much for [...]

Also works on Manjaro

plusline-morland, 2022-06-02:
> I've found that this does not work on Linux Mint 19.3 (Ubuntu 18.04): `LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-linux.AppImage`
> 
> Has this error: `[16186] Error loading Python lib '/tmp/.mount_UltimaxoHbXk/libpython3.10.so.1.0': dlopen: /lib/x86_64-linux-gnu/libm.so.6: version GLIBC_2.29 not found (required by /tmp/.mount_UltimaxoHbXk/libpython3.10.so.1.0) `
> 
> Per my issue #12344 mentioned above I have tried installing glibc-2.29 (per the above error message) alongside my 2.27 for the OS, but this is not successful: `LD_PRELOAD=/opt/glibc-2.29/lib/libc.so.6 ./Ultimaker-Cura-5.0.0-linux.AppImage` Has this error: `Segmentation fault (core dumped)`
> 
> Any thoughts?

Maybe read this note from the Releases:
"From version 5.0 onwards - Ultimaker Cura is only supported on operating systems actively maintained by their software manufacturer or community. This means Windows 7 and MacOS 10.14 will no longer be supported.
Ubuntu 18.04 is also no longer supported because of the update to Qt6."
Source: https://github.com/Ultimaker/Cura/releases

So you should upgrade your OS to something newer than Ubuntu 18.04.

leoheck, 2022-06-02:
Or continue using older versions of Cura, which is also fine.

[96 more replies]

PR #12736 (2022-07-12): Cura 9205 appimage not launching on ubuntu 2204
Building two AppImages, one based on Ubuntu 20.04 and one with on Ubuntu 22.04. This also solves the sign in button not working on some Linux systems. Keep in mind that this is more of an intermittent solution and that we should still try to solves this in a more generic/future proof-way

Fixes both CURA-9342 and CURA-9205

The results of this run can be found here: https://github.com/Ultimaker/Cura/actions/runs/2648505664

- Fixes #11856
- Fixes #12453
- Fixes #11900
- Fixes #12232
- Fixes #12291
commit 5e40f2ea (2022-07-12): Build an AppImage with the latest Ubuntu

Instead of trying to support multiple libstdc++ or ship or own (bad pratcice in general)
an easy solution could be to provide two different AppImage's, were one is build on the
latest Ubuntu runner.

This should fix #11856 for the 5.1.0 release, but I don't consider this a permanent and/or
sustainable solution. We should still create a ticket on the backlog to solves this issue
in a more generic way and future-proof way.

Contributes to CURA-9205
closed 2022-07-14 by jellespijker