Ultimaker/Cura #11858: Cannot access the Marketplace

csterlin, 2022-04-22:
### Application Version

5.0.0 Beta-1

### Platform

Windows 11, 11th Gen Intel(R) Core(TM) i7-11390H, 16.0 GB, Intel IRIS Xe

### Printer

Creality Ender 3

### Reproduction steps

1. Downloaded and installed application
2. Configured several profiles, copies of those installe by default.
3. Tried to use plugins, was unable to use any.
4. Tried to access the Marketplace, received notice that "Failed to load packages. Could not reach Marketplace. Retry?"
5. Retry Failed again.

### Actual results

Was unable to use the Marketplace.

### Expected results

The Marketplace should have opened with the requested plugins listed.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8536152/cura.log)
[CE3_Borderlands_Buggy_Thickened.zip](https://github.com/Ultimaker/Cura/files/8536157/CE3_Borderlands_Buggy_Thickened.zip)
[image]

Asterchades, 2022-04-22:
The plugins aren't maintained as part of the base Cura package - they're maintained separately by their individual developers (some of which happen to be part of the Cura dev team). 5.0 introduced a new UI framework (Qt6 instead of Qt5) and a fairly major overhaul of the slicing engine, so it will require those developers to update their plugins to become compatible again.

Evidently only the one plugin has been updated due to it being a case of simply changing one line in one file. Note that changing the same line in the same file for the other plugins doesn't work (I've tried) as it doesn't address the other compatibility issues.

This really isn't a bug.

nallath, 2022-04-22:
Well, the warning "Failed to load packages" shouldn't show, so that is a bug.

Asterchades, 2022-04-22:
Oh for sure. Though it's indeterminate (from an end-user point, at least) as to whether that error actually means it's having trouble communicating or if it's just reaching the end of the list and kicking back an error instead of the "No more results to load" banner. Checking the site-based marketplace does only show the same, lone plugin so I would lean more towards it being the latter.

FacuM, 2022-04-24:
Just wanted to point out that 4.x users **can't access the web marketplace** either!