Ultimaker/Cura #11859: Layer # in slicer preview does not match layer number in G-code. N+1 vs. N. Please make them match.

PhilBaz, 2022-04-22:
### Application Version

5.0

### Platform

win 10

### Printer

Voron 2.4

### Reproduction steps

Slice model...modify G-code based on preview. Change happens 1 layer later than expected based on preview.

### Actual results

Change happens 1 layer later than expected based on preview.

### Expected results

Sliced preview in Cura should have an initial layer number of 0 so it matches the G-code. People will understand. 

Preview and Gcode number should match for manual Gcode edits. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

DrewBatchelor, 2023-02-16:
I can +1, confirm this issue. in 5.1.1 and Win 11.

The Marlin GCode Cura outputs to the printer starts with layer zero. <- this is correct

The Graphical display Preview slicer bar in Cura starts at layer 1. - It should start at 0.
And if using any post processing such as "Pause at height", this happens one later later than requested.

GregValiant, 2026-09-07:
I'm cleaning up these old reports.  I will go ahead and close this.
If you use the post processor "Display Info on LCD" the printer screen (or a print server) will display the base 1 layer number while the Gcode will still use the Base 0 layer number.  All post processors that I have altered now use the preview layer number and make the translation to Base 0 in the background.

closed 2026-09-07 by GregValiant