Ultimaker/Cura #11862: Slicing error with Cura 5.0

KobeLAL, 2022-04-22:
### Application Version

5.0

### Platform

Windows 11

### Printer

Artillery Sidewinder X1

### Reproduction steps

Load the file to be printed into Cura. When I select my profile I get a slicing error. That is, the slicing does not work. 

### Actual results

I get an error message Slicing failed.

### Expected results

The slicing should be successful. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8537612/cura.log)

Vandresc, 2022-04-22:
Hi @KobeLAL, I can indeed reproduce as well on my Win 10. Will bring it up with the rest of the team. Thank you for your feedback.

PitbullRaven, 2022-04-22:
Windows 10. Same issue. 5.0.0 beta. 
Slicing failed with an unexpected error

The STL loads but doesn't slice.

Vandresc, 2022-04-25:
Hi @PitbullRaven, thank you, could you pls share the project file so we can make sure your case is fixed as well? Many thanks!

PitbullRaven, 2022-04-26:
Sure... Two files for basically the same thing. One a project file, just
try and load the model. The other is the STL. Both fail on me.

Best regards
Pitbull Raven

PitbullRaven, 2022-04-26:
@Vandresc 
[Files.zip](https://github.com/Ultimaker/Cura/files/8563281/Files.zip)

KobeLAL, 2022-04-26:
Interesting. I can load the individual STL and the slicing also works. If I open the project file but only click on Import models, the slicing does not work. If I load the complete project file, slicing does not work.  @PitbullRaven

Vandresc, 2022-04-26:
Devs,see CURA-9206 for our own internal reference.

PitbullRaven, 2022-04-26:
> Interesting. I can load the individual STL and the slicing also works. If I open the project file but only click on Import models, the slicing does not work. If I load the complete project file, slicing does not work. @PitbullRaven

I can conjure up other instances, if I am not mistake, where this same thing is happening.

https://youtu.be/czT1ww3YH5E

fvrmr, 2022-05-03:
This bug is fixed and will in the 5.0 stable release, so I will close this issue!