Ultimaker/Cura #11863: Cura 5.0 Beta does not start on Ubuntu 18.04

lokster, 2022-04-22:
### Application Version

5.0

### Platform

Ubuntu 18.04.6

### Printer

Custom FDM Printer

### Reproduction steps

Does not start on Ubuntu 18.04.6. Throws the following error:
[10691] Error loading Python lib '/tmp/.mount_UltimaVv58EP/libpython3.10.so.1.0': dlopen: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_UltimaVv58EP/libpython3.10.so.1.0)
Btw I have the same issue with the most recent stable release as well (4.13.1), so I use a modded version.

### Actual results

Does not run.

### Expected results

To run :)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

jellespijker, 2022-04-22:
@fvrmr did we updated the changelog with this information?

closed 2022-04-22 by jellespijker

lokster, 2022-04-22:
> The minimum supported version of Cura is Ubuntu 20.04 this is because our upgrade to Qt6.
> 
> we can no longer support Ubuntu 18.04 (EOL 2023) which has a glibc 2.27. Since we need at least a version 2.28 for glibc when compiling Qt6. This means we need drop support for 18.04 and make 20.04 (EOL 2030) out minimum supported version.

Sorry, but this is BS :) Just an excuse for lazy development, which in the end, will prevent atleast some users from upgrading. 
It won't matter how great Cura 5.0 is, if they can't run it.

VK1IAN, 2022-07-15:
Ubuntu 18.04 EOL is 2028, not 2023

jellespijker, 2022-07-15:
Correct I should have said EOS (end of support) instead of EOL.

But this unfortunately doesn't change the fact that we cannot support 18.04 due to our upgrade to PyQt6

lokster, 2022-07-15:
> Correct I should have said EOS (end of support) instead of EOL and EOS is 2023.
> 
> But this unfortunately doesn't change the fact that we cannot support 18.04 due to our upgrade to PyQt6

This is just a BS excuse. And I'm being polite here. The truth is that someone somewhere decided it will be cool to switch to Qt6, without thinking about backward compatibility. Because who cares... your main target are windows users, and they don't have a problem :)
What are the real benefits of switching to Qt6 for Cura? For the end user - not many. Almost identical interface, which in fact, looks even more buggy and inconsistent with the operating system's UI style & colors... Not to mention that from UX point of view some things don't make sense at all. But that's another story.

nallath, 2022-07-15:
> The truth is that someone somewhere decided it will be cool to switch to Qt6, without thinking about backward compatibility

If you're going to say that things are bullshit, it usually helps if your next sentence isn't bullshit as well ;) We didn't think it "would be cool to switch to Qt6" and we didn't ignore backwards compatibility. 

> What are the real benefits of switching to Qt6 for Cura? 

Support for M1 for apple. Faster interface, the ability to get future improvements in the mid/long run.

jellespijker, 2022-07-15:
@lokster also keep in mind that all of us devs are actually Linux orientated, we do our utmost best to keep it as  compatible as possible with older Linux systems, but sometimes a choice has to be made.

Now I'm sorry if you think we do that just be cool. Apart from the reasons @nallath provided to upgrade to Qt6 we have an additional reason why we can't support 18.04 anymore, since we switched to building our binaries with GH actions and the runners GH provide don't have an Ubuntu 18.04 instance. This additional upgrade was done because our old build system wasn't maintainable anymore; costed the yearly salary of a senior developer and was eating away tons of other resources (development time) just to patch things up. Time which now can be spend on adding value for user of Cura in terms of bugfixes and features.

lokster, 2022-07-15:
> Time which now can be spend on adding value for users of Cura in terms of bugfixes and features.

Sorry, but this is just hollow corporate/marketing speak for me i.e. Еxcuses instead of solutions. Fix this bug then. It's a legitimate bug :)
I loved Cura and used it almost daily for the past 5-6 years since v2. Was a big advocate for it in my small corner of the world and contributed one of the widely used, even if simple, plugins: Custom Supports. A feature which btw is STILL lacking in Cura after so many years of demand.
I can no longer pureheartedly do it (and I don't intend to), seeing that users like me are your least concern.
I am well aware that users like me are a minority, and I don't expect this will change anything.
So, for me, my frustration with this issue is anything but BS.

jellespijker, 2022-07-15:
> Еxcuses instead of solutions. Fix this bug then. It's a legitimate bug

Pull Request are welcome if you know of a solution which take all our stakeholders into account.

> I can no longer pureheartedly do it (and I don't intend to)

I'm sorry to hear that, I know a lot of users will be disappointed by that decision.

> seeing that users like me are your least concern.

As I said earlier **we** are user like you. Except that we maybe differ in assuming the reasons behind architectural choices.

VK1IAN, 2022-07-15:
> Sorry, but this is just hollow corporate/marketing speak for me i.e. Еxcuses instead of solutions. Fix this bug then. It's a legitimate bug :) I loved Cura and used it almost daily for the past 5-6 [...]

Let me know where your development skills end up and I will follow. I don't have the programming expertise, but am more than willing to help out where I can.

nallath, 2022-07-15:
> Sorry, but this is just hollow corporate/marketing speak for me i.e. Еxcuses instead of solutions. Fix this bug then. It's a legitimate bug

Correct, but it's simply too expensive to fix. As much as it sucks, there is a limit to the amount of resources that we can spend on things. I would have loved to be able to do it, but it's just not possible. We have to draw a line somewhere.

GKSeifert, 2022-09-19:
Given that you can't be bothered to support 18.04 the download page should not say simply "Linux 64 bit". It should specify Linux 64 bit at whatever level you do support. FYI, Freecad, the appimage downloaded yesterday at the most current level, runs just fine on my 18.04 system.

yzzyy5, 2022-09-28:
Had the same problem - can't use latest Cura on Ubuntu 18.04

Switched to PrusaSlicer. Their AppImage works fine. Same with FreeCAD and slic3r.  Their AppImage's all work fine.