Ultimaker/Cura #11864: [5.0 beta] Chrash when resizing windows within Cura

printingotb, 2022-04-22:
### Application Version

5.0 beta

### Platform

win10

### Printer

custom fff

### Reproduction steps

Click and hold while draging around the cursor for a while to rezise.
Seems to be every window that gets the diagonal mouse cursor. Print settings section does not produce a chrash.

### Actual results

Window freezes, drag cursor icon locks, chrashes after a couple of seconds.

### Expected results

to not chrash

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log.zip](https://github.com/Ultimaker/Cura/files/8538385/cura.log.zip)

fvrmr, 2022-04-26:
Hi @printingotb thank you for reporting.
I can't reproduce this myself. Does it happen all the time when you do this?

jake32321, 2022-06-01:
Seeing this same issue myself. Cura locks up after resizing the window multiple times. 

[image]
[cura.log](https://github.com/Ultimaker/Cura/files/8810904/cura.log)