Ultimaker/Cura #11865: [5.0 beta] Plugins "Uninstallation errors"

printingotb, 2022-04-22:
### Application Version

5.0 beta

### Platform

win10

### Printer

customfff

### Reproduction steps

Perhaps this is just a result of the plugins needing updates?

On my first run of 5.0 beta I choose the option to uninstall the incompatible plugins.
Now anytime I start Cura again, I get the error as shown in attached picture.

### Actual results

plugin uninstallation error

### Expected results

no error

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[cura.log.zip](https://github.com/Ultimaker/Cura/files/8538430/cura.log.zip)

fvrmr, 2022-04-25:
Hi @printingotb thank you for your report.
That shouldn't happen, will bring it up with the team. Keep you posted.

fvrmr, 2022-04-26:
I created a ticket for this on our backlog, devs see CURA-9202