Ultimaker/Cura #11866: [5.0 beta] Small inconvenience: Scrolling through the list of profiles much slower and less smooth than earlier versions.

printingotb, 2022-04-22:
### Application Version

5.0 beta

### Platform

win10

### Printer

custom fff

### Reproduction steps

Scroll the profiles dropdown

### Actual results

Scroll speed Cura 4.13.1 with my mouse settings is 5 rows per wheel step.
Scroll speed Cura 5 beta with my same mouse settings is instead averaged at 2.3 steps per row. So different amount of clicks per row.

### Expected results

Same scroll speed or at least a smooth scrolling experience.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no relevant additional info.

nallath, 2022-04-22:
One of our colleagues also reported this, so we already have an internal ticket number: CURA-9182

MIOsystems, 2022-05-20:
Just installed the Cura 5 release. But sorry, this is not a small inconvenience. It is unworkable for me. Scrolling trough every list in Cura is a pain in they ass. Scrolling needs to direct following the mouse-wheel. Not scroll further and stop at a random place. Place fix this!

Triangulix, 2022-05-28:
I experience the same issue. Cura is enforcing "smooth scrolling in lists", a "feature" that normally can be turned off in Windows system settings if a program doesn't enforce it.

ubershy, 2022-09-08:
I too experience the same issue. Navigating the settings is much harder than in Cura 4.

adripo, 2022-10-27:
Still happening. Print setting scrolling is almost unusable.
@nallath do you have any plan for a fix of the slow scrolling? Thanks

Benjamin4456, 2022-11-08:
How has this still not been fixed?  As of now I'm still on Cura 4.x because this one simple "feature" makes scrolling awful when using a mouse.  Not only does the page have inertia (like most folks are complaining about) but it also caps the scrolling speed way too low.  On a touchpad it works reasonably, but then the bounce-back becomes far too aggressive.

This is the sort of issue you expect to see on first time web developer sites where they think it "looks smooth", not in a developed program that had perfectly fine scrolling for ages.  Are there any plans for this to be fixed beyond "it's on an internal ticket"?

MIOsystems, 2022-11-09:
> My guess is that all Cura developers are using Mac or Linux and they don't care about the user experience on Windows. 

True. They love glitter and fanciness above functionality and hardcore engineering. Me already comment on more stupid fanciness within Cura. They aiming on morons that love a nice interface and single-click-print option. Please focus on the engineers!

ubershy, 2022-12-25:
It is not a small inconvenience for me as I use a lot of print settings. I'm still waiting for a fix...

Benjamin4456, 2022-12-25:
At this point I've jumped ship and moved to Super Slicer.  Honestly loving it so far with only a handful of things that I really miss from Cura - tree supports and some dual extrusion settings being the main ones.  I might try Cura again once this basic stuff gets fixed but I just can't deal with them putting off the glaringly obvious any further.  Cheers!

Triangulix, 2022-12-27:
I'm currently experimenting with Bamboo Studio. It looks more like Cura with the sidebar UI plug-in.

rick5295, 2023-03-19:
Almost a year later and this still has not been fixed. Please do something about this!!

mrodyhin, 2023-04-03:
agree, that's very annoying.

erentar, 2023-04-26:
Please add a toggle option to disable it

MIOsystems, 2023-05-05:
Dear developers. How can this issues still be in the latest release. It it a major UI bug. Please disable the smooth scrolling

midikeyboard, 2023-05-09:
this behaviour is unbearable... how is this still prevalent?

midikeyboard, 2023-05-09:
> > This is the sort of issue you expect to see on first time web developer sites where they think it "looks smooth", not in a developed program that had perfectly fine scrolling for ages. Are there any plans for this to be fixed beyond "it's on an internal ticket"?
> 
> My guess is that all Cura developers are using Mac or Linux and they don't care about the user experience on Windows. Maybe they even believe they can convert us to Mac or Linux by making the Windows user experience extra annoying.

I run Cura slicer in a NoVNC container and it's also present on ubuntu.

MIOsystems, 2023-05-09:
@midikeyboard, your guess is very good. That is exactly whats the case. Ultimaker GUI developers get very happy with glitter and fanciness. They think its commercially better i think. I think we need to fork Cura to get it back to a more engineer friendly design instead of office-guy-who-has-no-idea-how-printer-work design.

Triangulix, 2023-05-09:
> Dear developers. How can this issues still be in the latest release. It it a major UI bug. Please disable the smooth scrolling

They don't care.
BTW: Your comment has been hidden. Most of mine have been deleted. I have completely stopped reporting bugs to Cura and I can recommend to anyone to have a look at Bamboo Studio.

MIOsystems, 2023-05-10:
Please checkout #12272 to see a video impression of the effect it has on the user experience.

nallath, 2023-05-10:
> They don't care. BTW: Your comment has been hidden. Most of mine have been deleted. I have completely stopped reporting bugs to Cura and I can recommend to anyone to have a look at Bamboo Studio.

The comment has not been hidden. It's still visible.

Triangulix, 2023-05-11:
> The comment has not been hidden. It's still visible.

_This comment was marked as abuse._

nallath, 2023-05-11:
That one isn't. The one where remarks regarding the ability of people or purposeful ignoring of comments are marked as such. 

We do care about this software. But I don't care about being called names or insinuations that I (or my colleagues) are either incompetent or malicious. I also don't care for people calling novice users morons. Or do you feel that that is a good way to get issues resolved? I get that it's frustrating that an issue hasn't been solved, but we have limited resources to work on things. You're free to disagree with priorities that we set. If we see more (constructive) feedback that something needs to be changed, we will take that into account.

MIOsystems, 2023-05-11:
I wonder how these prioritizations are determent. If such thing as `Added the "CTRL +" shortcut to open Preferences window` is preferred above making simple GUI items work well, how can this issue ever become a priority?

nallath, 2023-05-11:
> is preferred above making simple GUI items work well

If you would have quoted the entire part of the changelog, you would have seen that it was a contribution ;) I can't say why other developers, that work for free, prioritize things. I'm just happy that they do <3