Ultimaker/Cura #11867: If retraction distance is set to less than 1mm on custom material, it will reset to zero.

Kumaloco, 2022-04-22:
### Application Version

Beta

### Platform

Windows 10

### Printer

Ender 3v2 modified

### Reproduction steps

1. Manage filament profile
2. Change retraction distance on custom material profile to less than 1mm
3. exit window
4.  check profile again

### Actual results

Retraction distance on profile is reset to zero

### Expected results

New retraction distance setting retained

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I tried setting the retraction distance for a material to less than 1mm, I'm using a Biqu h2 extruder, and when I viewed the material profile again it reset the distance to zero.

Asterchades, 2022-04-22:
It isn't just less than 1mm which isn't working - any value you enter in the material properties is being truncated. You can still get fractional values using the print settings.

See: https://github.com/Ultimaker/Cura/issues/11840

fvrmr, 2022-04-25:
Closing this issue since it is related to #11840

closed 2022-04-25 by fvrmr