Ultimaker/Cura #11868: Print settings explanation interface bugs

3Designing, 2022-04-22:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

CR10S

### Reproduction steps

1. sweep over print settings.
2. I open the Cura 5.0.0 and it informs me for that the version 5.0.0 is available yet I have the latest BETA version. For some reason it does not recognize that is up to date. 

### Actual results

The explanation for each setting appears as expected on the left of the print settings window.
The entire settings explanation is not fully visible, hence I tried to scroll down to see the rest.
But as I move my cursor from the print settings to the explanation, the explanation disappears.
I tried sweep over the specific print settings to the explanation but I was still unable to reach the window before it disappeared.

### Expected results

I was expecting to be able to reach the explanation window and scroll down from the slider to see the rest of it. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Not a major issue, but with so many settings to work with a fast explanation of each is very convenient.
Thanks all. Cura is a fantastic slicer and with the Arachne machine it will become far superior.

fvrmr, 2022-04-25:
Hi @3Designing thank you for your report. 
The tooltip issue is a known issue see: https://github.com/Ultimaker/Cura/issues/11834

About the version update, I noticed the same this morning. I will check with the team!

fvrmr, 2022-04-25:
Turn out is it a known issue, see: https://github.com/Ultimaker/Cura/issues/11229. We have a ticket on our backlog for this. 
Closing this issue since both bugs are duplicates.