Ultimaker/Cura #11869: [Cura 5.0 beta] Freeze when switching profile with active changes

printingotb, 2022-04-22:
### Application Version

[Cura 5.0 beta]

### Platform

win10

### Printer

custom fff

### Reproduction steps

Cura 5 imported all my profiles as expected and included the current changes to the active profile.
When I click another profile, the question to keep or discard changes opens blank and Cura immediately freezes.

However, if I choose discard changes BEFORE selecting a different profile there is no issue.

### Actual results

Cura freezes and needs to be restarted.

### Expected results

no freeze

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[cura.log (2).zip](https://github.com/Ultimaker/Cura/files/8538622/cura.log.2.zip)

fvrmr, 2022-04-26:
Hi @printingotb thank you for reporting.
Could you try to disable the USB printing plugin and see if that helps?

printingotb, 2022-04-26:
I have don't have that plugin.

Zazou49, 2022-05-01:
I have the same issue. I have dual monitor, can it be the issue ?

printingotb, 2022-05-01:
> I have the same issue. I have dual monitor, can it be the issue ?
It was for me:
https://github.com/Ultimaker/Cura/issues/11929

Zazou49, 2022-05-01:
I confirm, if I disable the second screen, everything works

closed 2022-05-23 by github-actions[bot]